mitchellspryn/UrdfSim #1: How to link control inputs to physics

sinanonur, 2019-04-04:
I was originally asking the same thing for the AirSim library but I was recommended to look at this repo. Thank you for working on such a thing.

My original question is here:
https://github.com/Microsoft/AirSim/issues/1864

I tried to read through the documentation you have provided. I am still not clear about how to map the control inputs to physics namely forces and torques on certain points. How can I define the kinematic model for my vehicle?

Thank you.

mitchellspryn, 2019-04-04:
It does depend on what type of vehicle you are trying to make. 

The code in this repo is good for making vehicles that can be thought of a sum of simple kinematic shapes. For example, a rover, an arm, etc. It can't do floating easily because that requires setting a bunch of settings in the editor (AFAIK there is no way to natively do "floating", you have to play games with special physics volumes). To set up a bot that, you'd follow the steps [here](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UsingAUrdfPawn.md). Roughly, the steps are:

1) Define your geometry in a xml file. This loosely follows the [ROS URDF](http://wiki.ros.org/urdf/Tutorials) specification, but there are lots of features that aren't supported. Look at the [examples](https://github.com/mitchellspryn/UrdfSim/tree/master/Examples/UrdfBot) provided for an idea of what to do.
2) Modify your settings.json to point to the xml. 
3) Use the python API to control the forces and torques. There are two ways of doing this. First, you can use AddForce() and AddTorque() to specify forces at a point. This is rather difficult to code, but gives you the most control. The other option is to use the built-in controllers like servo() and linear_actuator(). If your bot has components like this, then this will be the way to go as your simulation will be much more stable.

Some tips:

1) Try to make your URDF file as simple as possible. For example, rather than fusing together ten rectangles to make a frame, draw the frame in something like autocad and use the <mesh> element. This will make your simulation much more stable.
2) Use [debug mode](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UrdfDebugMode.md) if the URDF doesn't render as expected.

sinanonur, 2019-04-05:
> It does depend on what type of vehicle you are trying to make.
> 
> The code in this repo is good for making vehicles that can be thought of a sum of simple kinematic shapes. For example, a rover, an arm, etc. It can't do floating easily because that requires setting a bunch of settings in the editor (AFAIK there is no way to natively do "floating", you have to play games with special physics volumes). To set up a bot that, you'd follow the steps [here](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UsingAUrdfPawn.md). Roughly, the steps are:
> 
> 1. Define your geometry in a xml file. This loosely follows the [ROS URDF](http://wiki.ros.org/urdf/Tutorials) specification, but there are lots of features that aren't supported. Look at the [examples](https://github.com/mitchellspryn/UrdfSim/tree/master/Examples/UrdfBot) provided for an idea of what to do.
> 2. Modify your settings.json to point to the xml.
> 3. Use the python API to control the forces and torques. There are two ways of doing this. First, you can use AddForce() and AddTorque() to specify forces at a point. This is rather difficult to code, but gives you the most control. The other option is to use the built-in controllers like servo() and linear_actuator(). If your bot has components like this, then this will be the way to go as your simulation will be much more stable.
> 
> Some tips:
> 
> 1. Try to make your URDF file as simple as possible. For example, rather than fusing together ten rectangles to make a frame, draw the frame in something like autocad and use the  element. This will make your simulation much more stable.
> 2. Use [debug mode](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UrdfDebugMode.md) if the URDF doesn't render as expected.

Thank you for the quick response.

I'm still not clearon how I can addforce to a point using the python API. I couldn't find relevant documentation or example code. Can you point me to that direction.

Is this something specific to this repo or can I do that in Arsim as well.

Thank you very much.

mitchellspryn, 2019-04-05:
Check the [BoxAndCylinderClient](https://github.com/mitchellspryn/UrdfSim/blob/master/PythonClient/urdfbot/BoxAndCylinderClient.py). You're looking for the API calls of AddAngularForce(), AddLinearForce() and UpdateForceMagnitude(). 

I will reiterate that if you have servo / motor / linear actuator motion on your bot, you are better off using the controlled motion component, like in the [Lunabot client](https://github.com/mitchellspryn/UrdfSim/blob/master/PythonClient/urdfbot/LunabotClient.py). 

AFAIK you can't do this in AirSim.

closed 2019-04-05 by sinanonur