mitchellspryn/UrdfSim #2: UrdfSim not up to data with AirSim

nikolaif399, 2019-09-13:
AirSim has been updated since February, would it be possible to pull the latest AirSim into your UrdfSim fork? AirSim now uses settings.json 1.2, but UrdfSim is still on 1.0.

mitchellspryn, 2019-12-04:
Most likely no.

The projects have diverged, and doing a merge will be very messy. Given that most of the new features are for things that UrdfSim will not support (e.g. Unity support), I don't think it's worth the effort. Note that for the URDF, settings is completely different than for car / drone. 

Is there a particular feature that you are looking for? I could investigate cherry-picking it.

closed 2019-12-04 by mitchellspryn

ajberlier, 2020-01-26:
@mitchellspryn I am just looking to be able to drop a the urdf meshes into the most recent update of AirSim and control it through the API. There is not need to any of the Physics or any thing, just simply drop the mesh into AirSim with a Urdf file.

mitchellspryn, 2020-01-27:
I'm not sure what the question is here. 

Are you saying that you're using some external physics engine to compute the position of each of the different links? Then what's the point of using UrdfSim? I do not know your scenario.

Also, which specific features are you missing from the latest version of AirSim?