mitchellspryn/UrdfSim #3: uproject is not building

ajberlier, 2019-12-04:
The AirSim repo has a uproject in it and when it is built is add some dependencies. UrdfSim does not. How do I handle this?

When I build UrdfSim,
./setup.sh
./build.sh
There is not uproject or dependencies in UrdfSim/Unreal/Environments/Blocks.

Thank you for the help!

mitchellspryn, 2019-12-04:
There is no UProject because it's a plugin. 

You need to create your environment in Unreal. Then, copy-paste the plugin into the plugins directory and update the generated uproject to use it.

ajberlier, 2019-12-06:
I am building UrdfSim successfully it seems, but when I try to open my unreal project with the plugin copied over I get a message that UE4Editor-AirSim.dll is missing. When I build.cmd it does not create the .dll in plugins/binaries.

mitchellspryn, 2019-12-06:
That will be created when you build your unreal project.

Did you edit the UProject to include the new plugin? For reference, here are the steps I am currently using with success:

1) Build using build.cmd
2) Open Epic Games Launcher, create new project. 
3) Create new C++ class (to get the vs project generated for the first time). Doesn't matter what type of class. 
4) Close UE4. Navigate to the .uproject file on disk and make it look like this:

```
{
	"FileVersion": 3,
	"EngineAssociation": "4.18",
	"Category": "",
	"Description": "",
	"Modules": [
		{
			"Name": "YourProjectName",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
				"AirSim"
			]
		}
	],
	"Plugins": [
		{
			"Name": "AirSim",
			"Enabled": true
		}
	]
}
```
5) Create a "plugins" folder at the top level (in the same level as folders "saved", "source"). Copy the "plugins" folder from UrdfSim/Unreal into this folder, so you should have plugins/AirSim/AirSim.uplugin
6) Clear out the temporary files and generate clean. Delete the folders "binaries" and "intermediate", and the .sln file. Right click the .uproj and select "Generate visual studio project."

When you open in VS, you should be able to see that the plugin is now included in the source tree. Building (make sure you set the config to DebugGameEditor) should then build the plugin.

ajberlier, 2019-12-06:
I have been doing all but 6. I will try doing step 6 as well.

Just to note, I am able to run AirSim and make modifications without issue. When I copy and paste the UrdfSim plugins folder to my project I get the UE4Editor-AirSim.dll is missing.

build.cmd should generated the .dll in the Plugins/Binaries folder, correct?

Thank you for all the help! I just can nto seem to get past this error.

mitchellspryn, 2019-12-06:
I don't remember if build.cmd makes that dll. It should be irrelevant because it should be built as part of your uproj. 

Can you post your uproject file? Also, can you verify that you can see the source for UrdfSim inside of your solution like this:

[image]

ajberlier, 2019-12-06:
[image]

mitchellspryn, 2019-12-06:
See all those build errors at the bottom? The reason you're not finding the dll is because it's failing to compile. 

These are all issues introduced with VS 2017 15.9. You have a few options:

1) Use VS2015.  
2) Fix the compile errors. IIRC they were trivial syntax things
3) Wait until I get the fixes for 4.23 up.

-Mitchell

ajberlier, 2019-12-06:
Yes, I am just having trouble figuring out the root cause.
I am using DebugGameEditor. 
Are you using VS2015? I will try that to start. 
Do you have an expected completion date for the 4.23 release? 

Thank you for all the help!

mitchellspryn, 2019-12-06:
I'm guessing that [looking at the documentation for error C4596](https://docs.microsoft.com/en-us/cpp/error-messages/compiler-warnings/c4596?view=vs-2019) would help. I'm guessing there are a few technically unnecessary namespaces on the lines that are now giving warnings in vs2017. I do remember seeing those errors, and I think that's what I did to fix them.

I used 2015 when developing the version that is public.

I'm trying to go for the next few weeks (along with some custom created maps), but keep in mind that (unlike AirSim), this is completely separate from my current employment, so I can't promise on a delivery date.

mitchellspryn, 2019-12-06:
I took a look at the difference between what I have locally and what's in the repo. In the repo, I have 

```
static UrdfJointType UrdfJointSpecification::ParseJointType(FString type)
```

Locally, I've changed that to 
```
static UrdfJointType ParseJointType(FString type)
```

Rinse and repeat, and I think you'll be good to go.

ajberlier, 2019-12-06:
That all worked. Thank you. 

Now I am on to a new issue. If I run the unreal project with the settings.json set for the UrdfBot, unreal editor completely crashes. If I do nothing other than switch to the generic airsim settings.json, everything runs smoothly.

This explains what is happening to me but only when I switch to the urdf settings.json.
[https://github.com/Microsoft/AirSim/issues/84]
Thank you again for the help.

mitchellspryn, 2019-12-07:
What does your settings.json look like? As well as your URDF file? Keep in mind that the settings.json looks completely different for UrdfBot as opposed to car / drone.

ajberlier, 2019-12-07:
I am using the settings-lunabot.json renamed to just settings.json and with the paths matching the locations on my machine. https://github.com/mitchellspryn/UrdfSim/blob/master/Examples/UrdfBot/settings-lunabot.json

I am using the LunabotFromMesh.xml as the urdf with the paths matching the locations on my machine.
https://github.com/mitchellspryn/UrdfSim/blob/master/Examples/UrdfBot/LunabotFromMesh.xml

mitchellspryn, 2019-12-09:
Looking at the call stack, it looks like it's failing because the settings isn't a valid json. 

Unfortunately, I just tested this on my machine, and everything works as expected. Were any changes made to the code in this area? Can you check that the settings matches exactly what is in the repo and that it's a valid json (perhaps via [jsonlint](https://jsonlint.com/))?

Failing that, please post the json in its entirety here.

ajberlier, 2019-12-09:
That was a silly issue. I checked the jsonlint validation and it was valid, so I just recopied and pasted it over and it work fine.

But, I am now onto another issue. Possibly a pointer issue? I am trying to track it down now.

[image]

ajberlier, 2019-12-09:
I put the file content into jsonlint and it said it was valid. In a last ditch effort I changed the file path from "\" to "/" as it was in your settings.json and somehow that worked...

ajberlier, 2019-12-09:
I am now having the issue that it cannot find the .STL paths. I copy and paste the path that it claims it cannot find into windows explorer and it opens the file.

ajberlier, 2019-12-10:
I got it all figured out by debugging in visual studio. The issue was all "/" vs "\". I have been bouncing between windows and linux and had a mix of the two everywhere. Thank you for the help!

closed 2019-12-10 by ajberlier

mitchellspryn, 2019-12-17:
For the record. I've just pushed an update that will allow the project to build with UE4.24.

It also adds a few other API calls - notably for drawing things on the screen (e.g. spheres and lines) and for spawning/deleting objects from the scene.