mitchellspryn/UrdfSim #6: Should vhacd be vhacd_bucket in LunabotFromMesh.xml URDF?

ajberlier, 2019-12-10:
```
  <link name="top_ring">
    <visual>
      <geometry>                        <!--UPDATE THE PATHS TO POINT TO THE ABSOLUTE LOCATION OF THE STL FILES ON DISK-->
        <mesh type="stl_ascii" location="F:\SampleUrdf\LunabotParts\Frame3.STL" reverse_normals="true" scale_factor="100"/>
      </geometry>
      <material name="metal"/>
    </visual>
    <collision>
        <geometry>
            <mesh type="stl_ascii" location="F:\SampleUrdf\LunabotParts\Frame3.STL" reverse_normals="true" scale_factor="100" dynamic_collision_type="vhacd" vhacd_concavity="0.2" vhacd_output_folder_path="F:\**vhacd"**/>

```

mitchellspryn, 2019-12-10:
I don't think so.

What that parameter does: VHACD can be a time consuming process (not for these meshes, because they're simple, but for some other ones I've used it for it can take minutes / hours to run). Once the results are computed, they will be written into this folder. That way, for the next run, you can set the dynamic_collision_type to "manual" (I think), and skip the recomputation.

There are two parts that use this dynamic collision: the top frame and the bucket. They should be written to different folders. You can set the output paths to whatever you like, but the output files should not be mixed.

Is there a reason you think they should be the same?

ajberlier, 2019-12-11:
No, I just did not see a vhacd directory and did not know if it was supposed to create a new one. Thank you.

closed 2019-12-11 by ajberlier