mitchellspryn/UrdfSim #8: Was this tested on Linux?

ajberlier, 2020-01-21:
I am trying to get everything running on Ubuntu 18.04 but I am running into case issue and things that were not an issue for windows. For example, #include AirSimVehicle.h vs # include AirsimVehicle.h. Thank you!

mitchellspryn, 2020-01-21:
No, it was not tested on Linux. I probably won't have time to do so soon, because compiling isn't the only thing that you will need to do to get it up and running on another platform(IIRC packaging it on linux ran into problems linking MavLinkCom.lib when I tried it).

Perhaps running it in dockerized windows would be a quicker fix?

huckl3b3rry87, 2020-01-23:
I ran into the same issue today:/

@mitchellspryn are you suggesting to dockerize windows in linux? If so, [this is not possible](https://stackoverflow.com/questions/42158596/can-windows-containers-be-hosted-on-linux).

mitchellspryn, 2020-01-23:
My mistake. I thought docker would work. Looks like a full blown VM would be needed.

In any case, the answer to the original question is the same. Windows is the only supported platform for the forseeable future.

huckl3b3rry87, 2020-01-23:
@mitchellspryn thank you for the swift response.

Is this an issue with AirSim as well? Or is it just UrdfSim? Because it looks like the AirSim documentation says that [it works on Linux](https://microsoft.github.io/AirSim/docs/build_linux/), but to me this looks like an issue with AirSim

mitchellspryn, 2020-01-23:
I can't say for sure as I haven't run AirSim lately and I haven't attempted to run this codebase on linux to determine the exact causes of the error. 

Having said that, I know that linux compatibility was a big goal of AirSim, and I'd be surprised if it no longer worked on linux. I think they even had GCI builds set up for linux specifically. If I had to guess, it's probably an issue with the code specific to this project. The build processes for the two projects are largely the same, so it should be easy enough to check.

It's also worth noting that I haven't thoroughly tested the drone or car plugins with this codebase. I did a quick smoke test (i.e. they spawn and things look OK), but didn't go into the details (e.g. verifying output data). If you are intending to use the drone or car plugins, you are probably better off using AirSim. If you want to use the URDF builder, then you are better off using this repo.

huckl3b3rry87, 2020-01-23:
Thank you for the explanation:) 

I see, basically what I am looking for is a plugin that I can run that can simulate a gripper that picks up objects and then I will be collecting tactile data (maybe just points added to the a point could in the location where/if the gripper is grabbing the object) and RGBD data from a stationary camera. I have the camera part, but I still need the gripper and a way to generate tactile data ( I am a newbie to Unreal). 

For my needs, I don't need the URDF and a different gripper model would be fine for my case. For instance, I was planning on directly trying to use [this gripper](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/images/Arm.gif). Do you know if I could easily simulate a robotic arm for my use case with AirSim?

Unfortunately, I currently do not have a Windows Licence, but I am working on it. 

Thanks!

huckl3b3rry87, 2020-01-23:
I guess that I could use something like Gazebo, but I really like to avoid stringing together multiple simulators.

mitchellspryn, 2020-01-23:
No, AirSim won't work. AirSim only supports car and drone. 

Gazebo could work, but the sensor data would be lower quality. You may want to look into Nvidia Isaac if you have funding. 

Otherwise, I'd imagine the fixes for Linux would be mostly syntax, and should be pretty straightforward. Also, you'll need to learn some unreal, because you'd have to build your environment.

huckl3b3rry87, 2020-01-23:
Thank you!

I started to fix the build errors as @ajberlier noted, I fixed the capitalization issues with
```
#include AirSimVehicle.h
```

but currently I am stuck here
```
In file included from /home/hfebbo/Documents/workspace/Dataset_Synthesizer/Source 4.22_copy/Plugins/AirSim/Source/Vehicles/UrdfBot/SimModeUrdfBot.h:9:
/home/hfebbo/Documents/workspace/Dataset_Synthesizer/Source 4.22_copy/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfBotSimApi.h:9:10: fatal error: 'Airlib/include/physics/Kinematics.hpp' file not found
```
where I tried 
```
#include "../../Airlib/include/physics/Kinematics.hpp"
```
and
``` 
#include <Airlib/include/physics/Kinematics.hpp>
```
but neither change made the compiler find the file.

carmelodfr, 2020-05-07:
> Thank you!
>
> I started to fix the build errors as @ajberlier noted, I fixed the capitalization issues with
>
> ```
> #include AirSimVehicle.h
> ```
>
> but currently I am stuck here
>
> ```
> In file included from [...]

The issue, in this case, is Airlib that should be AirLib. the correct path is
> ```
> #include "AirLib/include/physics/Kinematics.hpp"
> ```
with capital L. There are a few other typos as mentioned before for
"vehicles/" that should be "Vehicles" and "AirsimVehicle.h" that should be "AirSimVehicle".

However, I still have many compilation errors at the moment ,e.g., 
 ```
StaticMeshGenerator.cpp:249:5: error: 'delete' applied to a pointer that was allocated with 'new[]'; did you mean 'delete[]'?
```

Has someone compiled urdfsim on linux successfully?

mtmal, 2020-06-02:
Hi there, you need to replace 'delete' with 'delete[]' as it's deleting an array.
I managed to compile it on Linux with UE 4.24, but with many headaches. 

First of all, there are many problems with wrong cases of files. Quoted example is just one from a few. You will also find errors in compilation where there is a code:
`if (...)
  if (...)
    statement A
  else
    statement B`
Else here is ambiguous, sou you will need to encapsulate the inner if-else with curly brackets.

Secondly, there is a problem with RTTI which prevent building UrdfSim UE project (could be same for AirSim). Some findings are reported here: [https://forums.unrealengine.com/development-discussion/c-gameplay-programming/1729429-issues-with-linux-plugin-development-ue4-24](https://forums.unrealengine.com/development-discussion/c-gameplay-programming/1729429-issues-with-linux-plugin-development-ue4-24). What I did was to set bUseRTTI to true in ModuleRules.cs and CppCompileEnvironment.cs and recompiled UE.

With the above, you should be able to build UE project under Linux and run some tests (e.g. HelloCar, although with HelloDrone the drone didn't lift off). I'm now looking at Urdf examples, as my plan is to parse my Mars-like rover into Urdf/UE.

Hope that helps! And sorry for digging out an old thread!

madmage, 2020-08-02:
Hi I am currently trying to build UrdfSim on a Linux machine. I am stuck at the problems already mentioned in this thread (when everything is working I would probably make a Pull Request to allow everybody to avoid these headaches :) ).

However, some of these errors were trivial to solve, I am stuck with a couple of them that require more knowledge about UrdfSim and Unreal:

`MyProject/Plugins/AirSim/Source/Vehicles/UrdfBot/MeshGeneration/StaticMeshGenerator.cpp:213:36: error: no matching member function for call to 'Compute'
    bool success = vhacdInterface->Compute(verts, numVerts, tris, numTris, VHACD_Params);
                   ~~~~~~~~~~~~~~~~^~~~~~~
/mnt/ssd/local2/urdfsim/UnrealEngine/Engine/Source/ThirdParty/VHACD/public/VHACD.h:91:18: note: candidate function not viable: requires 7 arguments, but 5 were provided
    virtual bool Compute(const float* const points,
                 ^
/mnt/ssd/local2/urdfsim/UnrealEngine/Engine/Source/ThirdParty/VHACD/public/VHACD.h:99:18: note: candidate function not viable: requires 7 arguments, but 5 were provided
    virtual bool Compute(const double* const points,
                 ^
`

and

`
In file included from MyProject/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:59:
MyProject/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfBotPawn.cpp:309:54: error: non-const lvalue reference to type 'FVector' cannot bind to a temporary of type 'FVector'
    this->root_component_->SetReferenceFrameLocation(this->GetActorLocation(), this->GetActorRotation());
                                                     ^~~~~~~~~~~~~~~~~~~~~~~~
MyProject/Plugins/AirSim/Source/Vehicles/UrdfBot/MeshGeneration/../UrdfLink.h:68:49: note: passing argument to parameter 'translation' here
        void SetReferenceFrameLocation(FVector &translation, FRotator &rotation);
                                                ^
`

madmage, 2020-08-02:
Sorry again, I saw there is already a PR to make it work on Linux, but it is still not merged.

Should we use 4.24, as it seems from here, or 4.18 (as I am currently doing), as explained in the README.md about the installation?

ssaigarimella, 2024-11-15:
@madmage @mtmal how do you create a blocks environment with the custom urdf in linux, with custom fork that compiles on linux?

ssaigarimella, 2024-11-18:
@mtmal Really appreciate your reply!