mitchellspryn/UrdfSim #11: Mecanum wheel

stevemartinov, 2020-04-08:
Hi,

Have you tried to simulate the mecanum wheel? How does UrdfSim work with such kinematics? Do I need to write my own math models for it?

mitchellspryn, 2020-04-08:
Mechanum wheels will be difficult. There are two ways you can consider approaching them. 

1) You can built the geometry for each roller and the hub in something like sketchup, import them (either via stl or fbx), and link each roller to the hub via a continuous joint. There would be no code changes needed. The problem with this approach will be that PhysX starts behaving weirdly when you have an object with many constraints. 

2) You can split open UrdfBotPawn.cpp and write your own c++ code to control the physics of the bot. This will not be trivial, especially if you want the animations to match up with the motion of the bot.

stevemartinov, 2020-04-08:
I actually do have an STL of the mecanum wheel. So you mean instead of attaching a simple cylinder, I need to attach every roller as continuous joint to the cylinder base wheel link (which is attached to the joint between itself and body)?

mitchellspryn, 2020-04-08:
Precisely.

mitchellspryn, 2020-04-10:
Given that this question appears to be answered, I'll close it. If there are more questions, feel free to reopen.

closed 2020-04-10 by mitchellspryn