mitchellspryn/UrdfSim #12: What files are changed?

stevemartinov, 2020-04-10:
I am thinking to pull the latest AirSim repo and apply UrdfSim specific changes to be able to use the latest AirSim repo. When I was browsing the code, I noted that the most of the files that need to be edited are in Unreal/Plugins/AirSim folder. What else should I note?

mitchellspryn, 2020-04-10:
That the two repos have diverged dramatically. 

Merging this repo into core AirSim isn't just a matter of adding the URDF-specific files. There have been a lot of (subtle) changes that have been done to almost the entire core stack to enable this scenario. An example of a few things that are different:

1) At the time, there were bugs related to sensors that were not positioned at the center of mass of the robot. They've been fixed in URDFSim. I imagine they've been fixed in AirSim at this point, but probably different. 

2) Speaking of sensors, note that all bodies in AirSim are rigid bodies, so there is no reason to support multiple sensors of the same type (e.g. two IMUs). In URDFSim, that is not the case - you could have an IMU attached to two different links. Changes were made to the sensor stack to support this. 

3) Coordinate systems. AirSim swaps between Unreal (left-handed NEU), NED (right handed NED), and Eigen (right handed NWU, I think) coordinates. This is because a lot of drone packages expect outputs in NED coordinates, so the conversion is necessary. For the URDF bot, we've stuck with unreal coordinates for the entire stack to make things simpler. 

There are many more gotchas, but I don't remember them all at the moment. I actually discussed merging this last year into core AirSim with the AirSim team last year. It turned out to be a much larger task than any of us were prepared to do at the time, and it's probably gotten worse as time goes on. There were also features that were missing from UrdfSim that would need to be taken care of to merge into AirSim (e.g. Unity support).

That being said, are there particular features that you are looking for? Depending on the complexity, I could look at backporting them from AirSim into UrdfSim.

stevemartinov, 2020-04-10:
Well to be honest, I have not tried yet the actual simulator. Yesterday I have finished of making my URDF with 9 rollers on each wheel (totaling 2100 lines of URDF code! I really hope PhysiX will not collapse).

To be honest, I have never coded on Windows and I hate this OS but because Unreal is better supported under that system, I decided to go with it. Unfortinately, I was not able to compile your version of code under VS 2019 and I had a lot of errors and digging 1 error created several others. I am pretty sure that is because I have a lack of understanding how this whole VS ecosystem works. Then I compiled the AirSim without any issues.

So that is the main reason I have started to port your code into the latest AirSim. Indeed, I could see the changes you had under sensors and Eigen coordinates which I have changed and compiled. I changed the whole AirSim with successful compilation and now going towards changing the Unreal folder. I will probably get some other non-compiling errors but at least I am getting somewhere.

Have you tried to compile your repo recently on the latest system with VS 2019 and Windows SDK 8.1?

stevemartinov, 2020-04-10:
To be more specific, I get this sort of error:

`AirLib.lib(client.obj) : error LNK2001: unresolved external symbol __CxxFrameHandler4 [C:\UrdfSim\DroneShell\DroneShell.vcxproj]`

mitchellspryn, 2020-04-10:
I'm using VS2017. Note that it appears that [Unreal doesn't support VS2019](https://docs.unrealengine.com/en-US/Programming/Development/VisualStudioSetup/index.html). 

Also, the latest version of the windows SDK appears to be [10](https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk/).

mitchellspryn, 2020-04-10:
I'm using VS2017. It looks like the latest version of the windows SDK is [this one](https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk/).

stevemartinov, 2020-04-10:
Yes, I will try to use VS2017

stevemartinov, 2020-04-10:
Yes I can confirm all works. Btw, where exactly do I need to specify settings.json to choose Urdf? Is it inside the Unreal project under AirSim plugins folder? Or directly under UrdfSim folder outside the Unreal?

stevemartinov, 2020-04-10:
Sorry, I now understood what their Documentation mean...

closed 2020-04-10 by stevemartinov

mitchellspryn, 2020-04-11:
For anyone finding this via google...what that means is that you must have "SimMode" set to "UrdfBot"  if you want the UrdfBot functionality.

ssaigarimella, 2024-11-13:
@mitchellspryn @stevemartinov What release/version of AirSim was UrdfSim forked from? I could not find a CHANGELOG.md in UrdfSim, which was in AirSim. I plan to merge the changes from AirSim version 1.3.1 into UrdfSim. Thanks