mitchellspryn/UrdfSim #13: Did not detect 3 elements

stevemartinov, 2020-04-10:
Hi,

When I tried to import my URDF, I get an error when starting UE:

`Error at startup: Error parsing 3-d vector from string , did not detect 3 elements.`

Do you have any idea what that could be?

mitchellspryn, 2020-04-11:
The error means what it says, that it tried to parse a 3-vector from your config file, but couldn't for some reason. 99% of the time, this means that there's a typo somewhere (e.g. separating them by comma instead of space).

Here are the two spots it can be thrown from:
[First](https://github.com/mitchellspryn/UrdfSim/blob/580aebce9903c5e71f69967f2877fe91cf569273/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/MeshGeneration/StaticMeshGenerator.cpp)

[Second](https://github.com/mitchellspryn/UrdfSim/blob/067a15bf4fb9d6b0c05f5e7e40c6112a47bb01dd/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfParser/UrdfParser.cpp).

If you can share the UrdfFile, I can take a look.

stevemartinov, 2020-04-11:
I have modified my URDF and no longer get that error. However, when I try to add at least 1 roller to the wheel, I get an exception:

`Exception thrown at 0x00007FF960927DBE (UE4Editor-AirSim-Win64-DebugGame.dll) in UE4Editor-Win64-DebugGame.exe: 0xC0000005: Access violation reading location 0x0000000000000008.`

My URDF looks like this (only base_link, 4 wheels and 1 roller):

```
<?xml version="1.0"?>
<robot name="physics">

  <material name="rubber">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>
  <material name="metal">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Metal/M_Metal.M_Metal'"/>
  </material>
    <material name="reflector">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>

  <link name="base_link">
    <visual>
      <geometry>                        <!--UPDATE THE PATHS TO POINT TO THE ABSOLUTE LOCATION OF THE STL FILES ON DISK-->
        <box size="0.5 0.3 0.1" />
      </geometry>
      <material name="metal"/>
    </visual>
    <collision>
        <geometry>
          <box size="0.5 0.3 0.1" />
        </geometry>
    </collision>
    <inertial>
      <mass value="20"/>
      <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
    </inertial>
  </link>
 
  
  <joint name="wheel_tl_joint" type="continuous">
    <parent link="base_link"/>
    <child link="wheel_tl_link"/>
    <origin rpy="0 0 0" xyz="0.25 0.18 -0.05"/>
    <axis rpy="0 0 0" xyz="0 1 0"/>
    <limit lower="0" upper="1.57079" effort="10000" velocity="1"/>
  </joint>
  
   <link name="wheel_tl_link">
    <visual>
      <origin xyz="0 0 0" rpy="1.57 0 0 " />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\roman\Documents\AirSim\GeniusURDF\meshes\wheel_left_reduced_ascii.stl" reverse_normals="true" scale_factor="0.01"/>
      </geometry>
      <material name="metal"/>
    </visual>
    <inertial>
      <mass value="0.66772"/>
      <inertia ixx="0.0007418" ixy="-1.4465E-07" ixz="-1.8717E-07" iyy="0.00074187" iyz="3.1606E-07" izz="0.00092662"/>
    </inertial>
     <collision>
       <geometry>
         <mesh type="stl_ascii" location="C:\Users\roman\Documents\AirSim\GeniusURDF\meshes\wheel_left_reduced_ascii.stl" reverse_normals="false" scale_factor="0.01" dynamic_collision_type="vhacd" vhacd_concavity="0.2" vhacd_output_folder_path="C:\vhacd"/>
       </geometry>
     </collision>
  </link>

  <joint name="wheel_tr_joint" type="continuous">
    <parent link="base_link"/>
    <child link="wheel_tr_link"/>
    <origin rpy="0 0 0" xyz="0.25 -0.18 -0.05"/>
[... 76 lines cut ...]
    <parent link="wheel_tl_link" />
    <child link="roller_tl_link_1" />
    <axis rpy="0 0 0" xyz="1 0 0" />
  </joint>

  <link name="roller_tl_link_1">
    <visual>
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\roman\Documents\AirSim\GeniusURDF\meshes\roller_reduced.stl" reverse_normals="true" scale_factor="0.01"/>
      </geometry>
      <material name="metal" />
    </visual>
    <inertial>
      <mass value="0.043693" />
      <inertia ixx="2.8636E-06" ixy="-3.7518E-16" ixz="4.7519E-15" iyy="8.7723E-06" iyz="-2.5223E-11" izz="8.7728E-06" />
    </inertial>
    <collision>
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\roman\Documents\AirSim\GeniusURDF\meshes\roller_reduced.stl" reverse_normals="true" scale_factor="0.01" dynamic_collision_type="vhacd" vhacd_concavity="0.2" vhacd_output_folder_path="C:\vhacd"/>
      </geometry>
    </collision>
  </link>

</robot>
```

stevemartinov, 2020-04-11:
I actually spotted the problem, it was because I did not define the "limit" for the roller joint.

I also want to ask whether there is an option to view the URDF first before pressing Play on UE4? Because with so many rollers, I cannot view properly how everything looks like and whenever I tick "Use controlRotation" everything collapses

stevemartinov, 2020-04-11:
Hi, I am still actively looking into this issue. I have uploaded my URDF (with Mesh included) and settings.json files. If you do not mind, can you try them? My STL has origin in the middle of my wheel but when I include that in URDF it does not seem to be centered. 

Here is my AirSim folder (the one in Documents folder):
https://drive.google.com/open?id=1Jv3yDzgHD6s7YJ9fssx5egdyDzoKKG50

stevemartinov, 2020-04-11:
I have updated my folder:
https://drive.google.com/open?id=1Jv3yDzgHD6s7YJ9fssx5egdyDzoKKG50

Now I am able to get all 4 wheels in a correct position but having troubles with the rollers...

mitchellspryn, 2020-04-12:
To answer your first question: your best bet on viewing it before importing it into UE4 is to use something like [rvis](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch). There isn't a great way to "preview" it via UrdfSim. 

I'm not going to have much time this weekend, but may have a few hours during the week next week if you are still having trouble. If you are having issues with orientation, enabling [debug mode](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UrdfDebugMode.md) may help you see the coordinate transformations.

stevemartinov, 2020-04-12:
I have noticed that the angles in URDF are different than in ROS. All angles go positive clockwise while in ROS it is anticlockwise and the Y direction is opposite to ROS. Can you point where can I change that in the code? I remember seeing this but can't remember where exactly

mitchellspryn, 2020-04-13:
Probably [UrdfBotPawn.cpp](https://github.com/mitchellspryn/UrdfSim/blob/23f2e5657cbac89ed558331d3186824854b2dac3/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfBotPawn.cpp)

koc-lab, 2021-03-06:
> 
> 
> I have modified my URDF and no longer get that error. However, when I try to add at least 1 roller to the wheel, I get an exception:
> 
> `Exception thrown at 0x00007FF960927DBE (UE4Editor-AirSim-Win64-DebugGame.dll) in UE4Editor-Win64-DebugGame.exe: 0xC0000005: Access violation reading location 0x0000000000000008.`
> (If I remove the roller, exception does not happen, so something has to do with attachment of the roller)
> 
> My URDF looks like this (only base_link, 4 wheels and 1 roller):
> 
> ```
> <?xml version="1.0"?>
> <robot name="physics">
> 
>   <material name="rubber">
>     <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
>   </material>
>   <material name="metal">
>     <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Metal/M_Metal.M_Metal'"/>
>   </material>
>     <material name="reflector">
>     <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
>   </material>
> 
>   <link name="base_link">
>     <visual>
>       <geometry>                        
>         <box size="0.5 0.3 0.1" />
>       </geometry>
>       <material name="metal"/>
>     </visual>
>     <collision>
>         <geometry>
>           <box size="0.5 0.3 0.1" />
>         </geometry>
>     </collision>
>     <inertial>
>       <mass value="20"/>
>       <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
>     </inertial>
>   </link>
>  
>   
>   <joint name="wheel_tl_joint" type="continuous">
>     <parent link="base_link"/>
>     <child link="wheel_tl_link"/>
>     <origin rpy="0 0 0" xyz="0.25 0.18 -0.05"/>
>     <axis rpy="0 0 0" xyz="0 1 0"/>
>     <limit lower="0" upper="1.57079" effort="10000" velocity="1"/>
>   </joint>
>   
>    <link name="wheel_tl_link">
>     <visual>
>       <origin xyz="0 0 0" rpy="1.57 0 0 " />
>       <geometry>
>         <mesh type="stl_ascii" location="C:\Users\roman\Documents\AirSim\GeniusURDF\meshes\wheel_left_reduced_ascii.stl" reverse_normals="true" scale_factor="0.01"/>
>       </geometry>
>       <material name="metal"/>
>     </visual>
>     <inertial>
>       <mass value="0.66772"/>
>       <inertia ixx="0.0007418" ixy="-1.4465E-07" ixz="-1.8717E-07" iyy="0.00074187" iyz="3.1606E-07" izz="0.00092662"/>
>     </inertial>
>      <collision>
>        <geometry>
>          <mesh type="stl_ascii" location="C:\Users\roman\Documents\AirSim\GeniusURDF\meshes\wheel_left_reduced_ascii.stl" reverse_normals="false" scale_factor="0.01" dynamic_collision_type="vhacd" vhacd_concavity="0.2" vhacd_output_folder_path="C:\vhacd"/>
>        </geometry>
>      </collision>
>   </link>
> 
>   <joint name="wheel_tr_joint" type="continuous">
[... 305 lines cut ...]
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\nurul\OneDrive\Belgeler\AirSim\LunabotParts\right_front_wheel.STL" reverse_normals="true" scale_factor="10"/>
      </geometry>
      <material name="rubber" />
    </visual>
    <collision>
      <origin xyz="0 0 0.0225" rpy="0 0 0" />
      <geometry>
        <cylinder length="0.045" radius="0.05" />
      </geometry>
    </collision>
  </link>

  <joint name="right_front_wheel_joint" type="continuous">
    <origin xyz="0 0 0" rpy="1.5708 0 0" />
    <parent link="right_steering_hinge" />
    <child link="right_front_wheel" />
    <axis xyz="0 0 -1" />
    <limit effort="10" velocity="100"/>
  </joint>
  
</robot>

nurullahsevim, 2021-03-06:
> 
> 
> Hi,
> 
> When I tried to import my URDF, I get an error when starting UE:
> 
> `Error at startup: Error parsing 3-d vector from string , did not detect 3 elements.`
> 
> Do you have any idea what that could be?
> 
> Debug message:
> 
> ```
> [2020.04.10-21.05.51:458][441]LogContentBrowser: Native class hierarchy updated for 'MovieSceneCapture' in 0.0008 seconds. Added 20 classes and 0 folders.
> [2020.04.10-21.05.51:474][441]Cmd: r.CustomDepth 3
> [2020.04.10-21.05.51:475][441]r.CustomDepth = "3"
> Exception thrown at 0x00007FFA3FDAA859 in UE4Editor-Win64-DebugGame.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00000036FBF6F6A8.
> Exception thrown at 0x00007FFA3FDAA859 in UE4Editor-Win64-DebugGame.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00000036FBF6FCD8.
> [2020.04.10-21.05.54:578][441]LogSlate: Window 'Error' being destroyed
> [2020.04.10-21.05.54:585][441]LogOutputDevice: Warning: 
> 
> Script Stack (0 frames):
> 
> Ensure condition failed: BeginPlayCallDepth - 1 == CurrentCallDepth [File:D:/Build/++UE4/Sync/Engine/Source/Runtime/Engine/Private/Actor.cpp] [Line: 3384] 
> 
> [2020.04.10-21.05.54:586][441]LogOutputDevice: Error: Ensure condition failed: BeginPlayCallDepth - 1 == CurrentCallDepth [File:D:/Build/++UE4/Sync/Engine/Source/Runtime/Engine/Private/Actor.cpp] [Line: 3384] 
> 
> 
> [2020.04.10-21.05.54:586][441]LogStats:             FDebug::EnsureFailed -  0.001 s
> UE4Editor-Win64-DebugGame.exe has triggered a breakpoint.
> ```

I have the same problem now, i.e. did not detect 3 elements. I wonder how you fixed it. I am sharing my xml file btw. Mitchell, can you check the file to see whether there is any problem with it.

<?xml version="1.0"?>
<robot name="physics">

  <material name="rubber">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>
  <material name="metal">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Metal/M_Metal.M_Metal'"/>
  </material>
    <material name="reflector">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>

  

  <link name="chassis">
    <inertial>
      <origin xyz="0.1477 0 0" rpy="0 0 0" />
      <mass value="4.0" />
      <inertia ixx="0.010609" ixy="0"          ixz="0"
                              iyy="0.050409"   iyz="0"
                                               izz="0.05865" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\nurul\OneDrive\Belgeler\AirSim\LunabotParts\chassis.STL" reverse_normals="true" scale_factor="10"/>
      </geometry>
      <material name="metal" />
    </visual>
  </link>

  

  <link name="left_rear_wheel">
    <inertial>
[... 158 lines cut ...]
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\nurul\OneDrive\Belgeler\AirSim\LunabotParts\right_front_wheel.STL" reverse_normals="true" scale_factor="10"/>
      </geometry>
      <material name="rubber" />
    </visual>
    <collision>
      <origin xyz="0 0 0.0225" rpy="0 0 0" />
      <geometry>
        <cylinder length="0.045" radius="0.05" />
      </geometry>
    </collision>
  </link>

  <joint name="right_front_wheel_joint" type="continuous">
    <origin xyz="0 0 0" rpy="1.5708 0 0" />
    <parent link="right_steering_hinge" />
    <child link="right_front_wheel" />
    <axis xyz="0 0 -1" />
    <limit effort="10" velocity="100"/>
  </joint>
  
</robot>

nurullahsevim, 2021-03-06:
ı have the same issue, i.e. did not detect 3 elements. Can you check the xml file below?

<?xml version="1.0"?>
<robot name="physics">

  <material name="rubber">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>
  <material name="metal">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Metal/M_Metal.M_Metal'"/>
  </material>
    <material name="reflector">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>

  

  <link name="chassis">
    <inertial>
      <origin xyz="0.1477 0 0" rpy="0 0 0" />
      <mass value="4.0" />
      <inertia ixx="0.010609" ixy="0"          ixz="0"
                              iyy="0.050409"   iyz="0"
                                               izz="0.05865" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\nurul\OneDrive\Belgeler\AirSim\LunabotParts\chassis.STL" reverse_normals="true" scale_factor="10"/>
      </geometry>
      <material name="metal" />
    </visual>
  </link>

  

  <link name="left_rear_wheel">
    <inertial>
      <origin xyz="0 0 -0.0225" rpy="0 0 0" />
      <mass value="0.34055" />
      <inertia ixx="0.00026046" ixy="0"          ixz="0"
                                iyy="0.00026046" iyz="0"
                                                 izz="0.00041226" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\nurul\OneDrive\Belgeler\AirSim\LunabotParts\left_rear_wheel.STL" reverse_normals="true" scale_factor="10"/>
      </geometry>
      <material name="rubber" />
    </visual>
    <collision>
      <origin xyz="0 0 -0.0225" rpy="0 0 0" />
      <geometry>
        <cylinder length="0.045" radius="0.05" />
      </geometry>
    </collision>
  </link>

  <joint name="left_rear_wheel_joint" type="continuous">
    <origin xyz="0 0.1 0" rpy="1.5708 0 0" />
    <parent link="chassis" />
    <child link="left_rear_wheel" />
    <axis xyz="0 0 -1" />
    <limit effort="10" velocity="100" />
  </joint>

  

  <link name="right_rear_wheel">
[... 126 lines cut ...]
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\nurul\OneDrive\Belgeler\AirSim\LunabotParts\right_front_wheel.STL" reverse_normals="true" scale_factor="10"/>
      </geometry>
      <material name="rubber" />
    </visual>
    <collision>
      <origin xyz="0 0 0.0225" rpy="0 0 0" />
      <geometry>
        <cylinder length="0.045" radius="0.05" />
      </geometry>
    </collision>
  </link>

  <joint name="right_front_wheel_joint" type="continuous">
    <origin xyz="0 0 0" rpy="1.5708 0 0" />
    <parent link="right_steering_hinge" />
    <child link="right_front_wheel" />
    <axis xyz="0 0 -1" />
    <limit effort="10" velocity="100"/>
  </joint>
  
</robot>

nurullahsevim, 2021-03-06:
I have the same issue. I append the xml code in a docx (i could not share in another way). Can you please check the xml code and tell if there is a mistake that caused this error? 
Thanks

nurullahsevim, 2021-03-06:
> I have modified my URDF and no longer get that error. However, when I try to add at least 1 roller to the wheel, I get an exception:
>
> `Exception thrown at 0x00007FF960927DBE [...]

Btw how could you fix your issue?

mitchellspryn, 2021-03-06:
So, if it works for the other four wheels, and not for the roller, I'm betting the problem is with the stl ascii file. 

STL importing is always a bit rough. The issue is that the engine has to dynamically compute the collision bounds, which can be a bit dicey, especially if the mesh isn't convex. There are a few things you can try:

* Try reducing the complexity of the the STL and seeing if it helps. You can import it into blender and decimate it. 
* Try importing it into the Unreal editor. I think you can just drag-and-drop .stl files. This will ultimately give you a better solution, as material properties can be baked in along with your asset. 

If you can get a better stack trace, then it would be easier to debug further.

nurullahsevim, 2021-03-06:
I don't know if I was clear about the problem. My problem is that I get "did not detect 3 elements" error. I am sorry if I was not clear. 
P.S.: cant drag-and-drop .stl files btw. I guess UE4 doesn't support it.

Thanks

mitchellspryn, 2021-03-06:
Try swapping "rpy" and "xyz" in the <origin> tag of the link. There was a report earlier that this caused a crash for some reason. 

If UE4 doesn't support STL, try importing to blender and exporting in .fbx. That should definitely work.

nurullahsevim, 2021-03-06:
I transformed my .stl files to .fbx and dragged into unreal. Now, I use them as unreal mesh type. However, I still get the same error although I also decimated the meshes. How can I track the error and find out what actually causes this problem? The final xml code is appended.
[xml.txt](https://github.com/mitchellspryn/UrdfSim/files/6096022/xml.txt)

mitchellspryn, 2021-03-06:
If you're running from within visual studio, you should get a stack trace in the output when it fails. That should point to where the problem is. 

If you're trying to use precompiled binaries, try building from source and including it in a project.

henry-tra, 2021-04-01:
> I transformed my .stl files to .fbx and dragged into unreal. Now, I use them as unreal mesh type. However, I still get the same error although I also decimated the meshes. How can I track the error [...]

try to add rpy="0 0 0" to the axis at the joint section