mitchellspryn/UrdfSim #14: Strange interaction between two links

stevemartinov, 2020-04-12:
Hi,

As you already know, I am simulating a mecanum robot. I make a continious joint between the wheel and the roller and applied vhcad collision to both wheel and roller links.

However, when I move around, I can see that when roller is happened to be a point of contact with a ground, it kind of seems to be "soft", meaning that it goes a bit inside the wheel. How can I prevent that?

stevemartinov, 2020-04-12:
Also I think I found the problem with rotated joints. As the roller has continuous joins which are not at 0 degrees (from 45 degrees) it means that the joints should rotated from its respective axis. However, when I turned on the debug mode, it showed me that the roller joint is rotating at the same angle as the wheel itself (at 0 degrees) and not at angled degree and when the joint starts to rotate at the same angle as the wheel, you can see that the roller starts to rotate weirdly.

Do you think that this is a possible bug?

stevemartinov, 2020-04-12:
Here is the video after I inserted all the rollers and made their joints to be fixed. As you can see, when I try to turn the robot in place (by giving left side + direction and the right side - direction), something strange happens. Do you have any idea why?

https://drive.google.com/file/d/1iPGw_hN9eXb-s-nro175la-QiJFnNl8y/view?usp=sharing

mitchellspryn, 2020-04-13:
Yeah...this was what I was afraid of...

While this very well may be a bug, I saw this happen when I tested a complex bot that had a lot of different parts. The bouncing you see is because PhysX doesn't solve the kinematics of each object with a high enough degree of accuracy. The errors accumulate, and eventually you start seeing artifacts like in the video. 

Unfortunately, the only real way to solve this would be either by switching to [actuations](https://documentation.help/NVIDIA-PhysX-SDK-Guide/Joints.html) or to a completely different engine entirely. 

Can you show me a video with debug mode on? I don't quite understand your axis comment.

Also, when running, have you given the viewport focus? I.e. you have clicked on the viewport. Unreal will slow down dramatically if you have focus on something else (e.g. a console window).

stevemartinov, 2020-04-14:
I have looked through the actuations but there are no continuous joints, is that correct? Or did you mean to use D6? Also can you point where do you create joints which are movable?
I will send the video later

mitchellspryn, 2020-04-14:
The supported joint types are in [UrdfJointSpecification.h](https://github.com/mitchellspryn/UrdfSim/blob/580aebce9903c5e71f69967f2877fe91cf569273/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfParser/UrdfJointSpecification.h). The definitions match the [Urdf Specification](http://wiki.ros.org/urdf/XML/joint) as far as I'm aware. I would think that for the rollers, you'd use a continuous joint, which allows it to roll around one axis. 

When I was talking about actuations, what I meant is that in PhysX, the concepts of "joints" and "actuations" are actually two separate things. Joints are more performant but less accurate. AFAIK using the unreal framework, joints are exposed on actors, but not actuations (because they're newer). 

If you want an example of using continuous joints, take a look at the [Lunabot example](https://github.com/mitchellspryn/UrdfSim/blob/580aebce9903c5e71f69967f2877fe91cf569273/Examples/UrdfBot/LunabotFromMesh.xml). This is the specification for the top GIF in the main README.

stevemartinov, 2020-04-15:
Do you think that switching to another engine like Bullet: https://github.com/bulletphysics/bullet3
would be easier than just chaning your URDF code to Nvidia Custom joints?

mitchellspryn, 2020-04-15:
It's a good thought. This idea was actually explored a long time ago when we were initially designing AirSim. However, it turned out to be difficult to use an external physics engine. IIRC, the primary complication was synchronizing the clock between the two engines. 

If we wanted to go this route, probably [Mujoco](http://www.mujoco.org/) would be the best physics simulator to use.