mitchellspryn/UrdfSim #15: unreal block environment

Jianghn-dev, 2020-04-21:
hi, I want to know that why is not having .uproject file in the unreal block environment .

thanks

mitchellspryn, 2020-04-21:
You might need to run [GenerateProjectFiles.bat](https://github.com/mitchellspryn/UrdfSim/blob/master/Unreal/Environments/Blocks/GenerateProjectFiles.bat).

However, note that the blocks environment is **unsupported** and may not run. The intent is for you to include this plugin in a project of your own.

Jianghn-dev, 2020-04-21:
yes,I build my own project finally. 
I still have tow questions.
First,Isn't urdfsim building on linux so far? I try ，always tip “#include Vehicles/AirsimVehicle.h” file not found.
Second，how to adjust perspective in airsim? is as below？            
 "Cameras": {
                "Follow": {
                    "X": -300, 
                    "Y": 30, 
                    "Z": 200,
                    "Pitch": -30, 
                    "Roll": 0, 
                    "Yaw": 0,
                    "AttachLink": "base"
                },
these parameters is adjusted by what principle

mitchellspryn, 2020-04-21:
1) [No](https://github.com/mitchellspryn/UrdfSim/issues/8).

2) I guess by "perspective" you mean the position of the camera following the bot relative to the location of the bot. If so, yes, that is where you adjust it. The "x, y, z" represent the x, y, and z offsets from the link to which you attach. It will remain fixed at this relative location, so if that link moves, the camera will move with it.

Jianghn-dev, 2020-04-21:
thank you for your reply.
when displaying urdf model, it display red or blue axis and line. How to remove redundant displays？

mitchellspryn, 2020-04-21:
Do you have [debug mode](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UrdfDebugMode.md) enabled? If so, turn it off. 

If not, please post a picture, as I'm not sure what lines you are referring to.

Jianghn-dev, 2020-04-23:
yes,I turn debug mode of,then it is ok.
I find urdfsim support unreal mesh，but how is unreal mesh UnrealMesh.xml  writed?

Jianghn-dev, 2020-04-23:
yes,I turn debug mode off,then it is ok.
I find urdfsim supports unreal mesh，but how is unreal mesh UnrealMesh.xml writed? And, in the example file it is not found.

Jianghn-dev, 2020-04-23:
Yes,I turn debug mode off,then it is ok.
I find urdfsim supports unreal mesh，but how is unreal mesh UnrealMesh.xml writed? And, in the example file it is not found.
The current urdfsim is based on which version of AirSim.

mitchellspryn, 2020-04-23:
I don't understand your first question. If you're asking about the 'mesh type="stl_ascii"' tags in the [Lunabot Example](https://github.com/mitchellspryn/UrdfSim/blob/master/Examples/UrdfBot/LunabotFromMesh.xml), then you have to modify the paths in the XML files to point to the absolute paths to where you have [the stl files](https://github.com/mitchellspryn/UrdfSim/tree/master/Examples/UrdfBot/LunabotParts) downloaded. 

I'm not sure the exact timestamp when the codebases diverged, but I think it was around November last year.

Jianghn-dev, 2020-04-23:
first question, I found these as below in your doc
If using a static mesh built inside the Unreal Editor, use the same mesh as a custom mesh, but specify a mesh type of 'unreal_mesh', and set the location to the full path of the mesh. This link should have no collision element - the simple collision of the mesh will be used automaticall (complex collision cannot be used, as physics must be simulated). An example can be seen in the UnrealMesh XML in the examples folder.
in fact, I don't find UnrealMesh XML in the examples folder.

mitchellspryn, 2020-04-23:
Oh. I see. 

That is for if you design an mesh inside of unreal (i.e. an asset that inherits from StaticMeshComponent). Then, this parameter should point to the [asset path](https://docs.unrealengine.com/en-US/Engine/Basics/AssetsAndPackages/index.html). This is not to be confused with the STL mesh, which has an external file.

Jianghn-dev, 2020-04-23:
[image]
i write xml as below,but it is error like picture
```
    <visual>
      <geometry>
        <mesh type="unreal_mesh" location="C:/Users/Administrator/Desktop/Unreal Project/Block/Blocks/Content/dj1.uasset"/>
      </geometry>
      <material name="metal"/>
    </visual>
```

mitchellspryn, 2020-04-23:
Right, like I said before...you don't use the path on disk when using the unreal mesh. You use the asset path. It would look something like this:

<mesh type="unreal_mesh" location="StaticMesh'/Game/wheel.wheel'" reverse_normals="true" scale_factor="10"/>

Jianghn-dev, 2020-04-23:
[image]
em..now, after clicking on play button, it has also error.

Jianghn-dev, 2020-04-23:
em..now, after clicking on play button, it has also error.
[image]

Jianghn-dev, 2020-04-23:
em..now, after clicking on play button, it has also error. UE4 crash reporter
[image]

Jianghn-dev, 2020-04-23:
sorry, I still have question about material setting. It is my xml ,but i found material likely is ignored .
```
<?xml version="1.0"?>

<robot name="physics">

  <material name="rubber">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/M_Reflector.M_Reflector'"/>
  </material>
  <material name="metal">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Metal/M_Metal.M_Metal'"/>
  </material>
  <material name="reflector">
    <unreal_material path="Material'/AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Reflector/MI_Reflector_Orange.MI_Reflector_Orange'"/>
  </material>
  
  <link name="base">
    <visual>
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\Administrator\Documents\AirSim\UrdfBot\box.stl" reverse_normals="true" scale_factor="100"/>
      </geometry>
      <material name="reflector"/>
    </visual>
    
    <collision>
      <geometry>
        <mesh type="stl_ascii" location="C:\Users\Administrator\Documents\AirSim\UrdfBot\box.stl" reverse_normals="true" scale_factor="100" dynamic_collision_type="vhacd" vhacd_concavity="0.2" vhacd_output_folder_path="C:\Users\Administrator\Documents\AirSim\UrdfBot\LunabotParts\vhacd"/>
      </geometry>
    </collision> 
    
    <inertial>
      <mass value="1.477" />
      <origin xyz="0 0 0" />
      <inertia ixx="0.01152" ixy="0.0" ixz="0.0" iyy="0.01152" iyz="0.0" izz="0.0218" />
    </inertial>
  </link>

</robot>
```
[image]

mitchellspryn, 2020-04-23:
I'm not entirely sure what's going on with the stack trace you posted, are you still seeing that error?

Do those materials exist at the paths at which you are referencing them in your unreal project?

Jianghn-dev, 2020-04-24:
About materials, i use your lunabot model, the materials setting is ok as below.And i use simple box shape, it aslo is ok. But if using quad hector model like above picture, it isn't ok.
[image]

Jianghn-dev, 2020-04-24:
[image]
About materials, i use your lunabot model, the materials setting is ok as below.And i use simple box shape, it aslo is ok. But if using quad hector model like above picture, it isn't ok.

Jianghn-dev, 2020-04-24:
I found again that if using a custom component stl, material setting is likely ignored. Only simple shapes can be set material(i.e.cylinder、box、ball). I don't know if I set it up wrong.

mitchellspryn, 2020-04-28:
That's interesting. In theory, if it works for standard shape (box, ball, etc), it should work for custom shape. They use the same code path. 

It makes me wonder if something is getting messed up when reading your STL. Looking at what you've posted above, it looks like it's floating in mid-air. Is collision generated? You can test this by running your project and using the console 'pxvis collision' to see if it's been generated properly. 

If you can share your stl, I can take a look.

Jianghn-dev, 2020-04-30:
My STL model isn't floating in mid-air, when clicking on the playerbutton, it lands ground. I think collsion is generated.  And, above picture, lunabot model seems like that material setting except bottom four cylinders is ignored. This is my STL.
[quadrotor.stl.zip](https://github.com/mitchellspryn/UrdfSim/files/4556733/quadrotor.stl.zip)

mitchellspryn, 2020-05-29:
Sorry, just getting time to look at this issue. 

It looks like your quadrotor has disconnected parts. I'm not sure how the meshing algorithm will handle that, as it probably expects everything to be 1 contiguous surface. 

As a more meta point, I don't think you should be using UrdfSim for quadroters. Presumably, you want it to fly. UrdfSim doesn't handle flying / buoyancy. I'd recommend looking at [AirSim](https://github.com/microsoft/AirSim). In particular, I know it's possible to create your own drone class, which will handle the physics (more) correctly.