mitchellspryn/UrdfSim #16: Error at startup

MiPa12, 2020-05-09:
Hi, 
I try to set up your plugin. I got this error when I run the unreal simulation: 
Error at startup: Unable to open file F:/vhacd/header.txt. 
I haven't found the folder vhacd. I have only found this one: C:\Users\Administrator\UrdfSim\Examples\UrdfBot\LunabotParts\vhacd_bucket\header.txt. 
I have changed the paths in settings-lunabot.json file and LunabotFromMesh.xml. Should I change paths somewhere else? 
best, Michael

mitchellspryn, 2020-05-09:
That error message is probably coming from the [static mesh generator](https://github.com/mitchellspryn/UrdfSim/blob/580aebce9903c5e71f69967f2877fe91cf569273/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/MeshGeneration/StaticMeshGenerator.cpp). It probably means that you are trying to read in a mesh in "manual" mode, but haven't actually generated the mesh yet. 

The doc you want is [this one](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/CustomMesh.md). Most likely, you will need to change from "manual" to "VHACD" mesh generation somewhere in your URDF file.

MiPa12, 2020-05-15:
I'm using the example URDF: LunabotFromMesh.xml. I'm not sure where I should change this. Can you explain shortly how to set up the lunabot example correctly. I copied the generated Plugin folder from UrdfSim in my unreal project folder. Pasted the content from settings-lunabot.json in the settings.json file. Changed the paths in the settings.json file to the location of the LunabotFromMesh.xml. Furthermore, I changed the paths from LunabotFromMesh.xml to the location of the STL files. I tested it with Visual Studio 2017. Switchted the Game Mode to AirSimGameMode. My uproject file looks like that: {
    "FileVersion": 3,
    "EngineAssociation": "4.24",
    "Category": "Samples",
    "Description": "",
    "Modules": [
        {
            "Name": "UrdfSimTest",
            "Type": "Runtime",
            "LoadingPhase": "Default",
            "AdditionalDependencies": [
                "AirSim"
            ]
        }
    ],
    "TargetPlatforms": [
        "MacNoEditor",
        "WindowsNoEditor"
    ],
    "Plugins": [
        {
            "Name": "AirSim",
            "Enabled": true
        }
    ]
}

Is there something wrong or missing? any tip will be appreciated, thank you

mitchellspryn, 2020-05-15:
Can you show your full LunabotMesh? It's likely that the problem is there.

guldamkwak3114, 2021-02-15:
I have the same error. Do you have any solution for this?