mitchellspryn/UrdfSim #17: AirSim version v1.3.1 with Visual Studio 2019

MiPa12, 2020-05-09:
Hi, 
I tried to use this plugin with the current AirSim version which does not support Visual Studio 2017 anymore (only VS 2019). Are you planning to do an update ? Would be great. 
Best, Michael

mitchellspryn, 2020-05-09:
Odd..I am using it with VS2019 on my box. This should not be used with AirSim, but in place of it. 

There are currently no plans to merge this into AirSim, or to pull the latest AirSim into this project. See [this related issue](https://github.com/mitchellspryn/UrdfSim/issues/2).

Ootang2019, 2020-05-14:
hi mitchellspryn,
is there any progress on this discussion? I am also trying to build UrdfSim with VS2019. 

I am new to AirSim and programming on Windows10. Recently, I have tried _build.cmd_ in _Developer_Command_Prompt_for_VS_2019_. It says it is not able to build with vs2019. Then, I tried modify _build.cmd_ by replacing this line _cmake -G"Visual Studio 15 2017"_ with _cmake -G"Visual Studio 16 2019"_. But it is still not working. Is there anyway to build with vs2019 or I have to delete it and reinstall with vs2017? 

any tip will be appreciated, thank you

mitchellspryn, 2020-05-29:
My comments [here](https://github.com/mitchellspryn/UrdfSim/issues/2) are relevant. No, I have not kept this up to date with the latest AirSim, and probably won't realistically have time to do it anytime soon. Try installing VS2017.

cthoma20-arc, 2020-07-01:
Hey @Ootang2019, were you able to figure out how to get it to work with Visual Studio 2019? I tried similar steps, and I'd prefer not to have to use VS 2017 if possible.

jpapon, 2020-07-23:
Making progress on building from scratch using 2019 on a fresh install.
You'll need to modify the line specifying 2017 in build.cmd to `cmake -G"Visual Studio 16 2019" ..`
You also need to make sure you install the 2017 C++ build tools.

Currently stuck on a ton of linker errors like this:
AirLib.lib(client.obj) : error LNK2019: unresolved external symbol __std_system_error_allocate_message referenced in function "public: __cdecl std::_System_error_message
::_System_error_message(unsigned long)" (??0_System_error_message@std@@QEAA@K@Z) [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(client.obj) : error LNK2019: unresolved external symbol __std_system_error_deallocate_message referenced in function "public: __cdecl std::_System_error_messa
ge::~_System_error_message(void)" (??1_System_error_message@std@@QEAA@XZ) [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(client.obj) : error LNK2001: unresolved external symbol __CxxFrameHandler4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(rpc_error.obj) : error LNK2001: unresolved external symbol __CxxFrameHandler4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(response.obj) : error LNK2001: unresolved external symbol __CxxFrameHandler4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(format.obj) : error LNK2001: unresolved external symbol __CxxFrameHandler4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(client.obj) : error LNK2001: unresolved external symbol __GSHandlerCheck_EH4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(rpc_error.obj) : error LNK2001: unresolved external symbol __GSHandlerCheck_EH4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
AirLib.lib(format.obj) : error LNK2001: unresolved external symbol __GSHandlerCheck_EH4 [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]
c:\workspace\UrdfSim\DroneShell\build\x64\Debug\DroneShell.exe : fatal error LNK1120: 4 unresolved externals [c:\workspace\UrdfSim\DroneShell\DroneShell.vcxproj]

jpapon, 2020-07-23:
Okay, got it to build. I had to retarget everything to 2019 and then disable WX because I had a few warnings, but after that, build finished without issue.

CYLOH3898, 2020-10-15:
> Okay, got it to build. I had to retarget everything to 2019 and then disable WX because I had a few warnings, but after that, build finished without issue.
Hi @jpapon , Can I know exactly how can I retarget everything to 2019 and disable WX? Thank You as I am quite new in UrdfSim