mitchellspryn/UrdfSim #19: Issue spawning URDF bot

huckl3b3rry87, 2020-05-12:
I followed the tutorials for AirSim and URDF sim, but I am having an issue trying to spawn a URDF bot into Unreal. 

It seems that there are no Blueprints for the URDF model, so I tried to simply drag and drop it from the Window in the picture into the Unreal editor, but nothing seems to show up

[image]

Sorry, I am a newbie to Unreal. Thank you for any help or advice

mitchellspryn, 2020-05-12:
The bot is a C++ class, not a blueprint class. All you should have to do is drag a player start into the level and set the game mode to "AirSimGameMode." Then, assuming that your URDF file / settings are correct, it should spawn.

huckl3b3rry87, 2020-05-12:
@mitchellspryn thank you for your patience and quick explanation:) Your suggestion worked!wwwwwwwwwwwwwwwwwww

closed 2020-05-12 by huckl3b3rry87