mitchellspryn/UrdfSim #20: Crashing when hit play linearactuator.cpp:91 and

huckl3b3rry87, 2020-05-14:
When I try to insert a URDF into a new map, it is crashing with on these lines of code
```
Unhandled exception

KERNELBASE
VCRUNTIME140
UE4Editor_AirSim!LinearActuator::ComputeForces() [c:\users\admin\documents\workspace\dataset_synthesizer - copy\source\plugins\airsim\source\vehicles\urdfbot\controlledmotioncomponent\linearactuator.cpp:91]
UE4Editor_AirSim!AUrdfBotPawn::Tick() [c:\users\admin\documents\workspace\dataset_synthesizer - copy\source\plugins\airsim\source\vehicles\urdfbot\urdfbotpawn.cpp:117]
```
For the first one it looks like it is getting a control signal ``noopControlSignal > 1.0f`` I did not write a [client to simulate the the arm](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UrdfClient.md), for now, I am just trying to insert it into my simulation.

It works when I do not have a map, but when I put it in a room with stuff this issue pops up.

mitchellspryn, 2020-05-15:
I don't understand what you mean "Works when I do not have a map."

When I look at that line in the code, I see 
```cpp
throw std::runtime_error("The initial state of the actuator could not be resolved. The actuator is extended longer than the maximal allowable distance.");
```

This indicates that the state of your actuator is outside the range of motion specified in your URDF file.

huckl3b3rry87, 2020-05-15:
@mitchellspryn thank you for the response.

Sorry, I misspoke. In both cases, I do have a map. But, when I try to only change the Project Map, under Project Settings -> Project -> Maps & Modes -> Default Maps -> Game Default Map, to something more than a very simple map,  for instance an elaborate room, it was crashing when I started.

I think that maybe it was hitting something as it fell down during player start and put it into an impossible state.

But, I just retried it far away from other objects and it works !

Thanks!

mitchellspryn, 2020-05-15:
Ah. I see. 

Perhaps, but also I've noticed that the simulation can be a bit unstable right at the start of play, especially when (A) playing in the editor, and (B) when not starting on the ground. This does tend to fix itself if the game is exported.

huckl3b3rry87, 2020-05-20:
Hmm, I am still running into this issue. Here is another example

If the player start looks like this 
[image]

it loads fine
[image]

* also it loads fine turned to the other side and back. 

But if I try to turn the gripper towards the camera, so that it can pick up the cup

[image]

It crashes with the above error.

Also, I tried to move the player close to the ground, but the error still exists. Is there a way to move the player directly on the ground?

closed 2020-05-21 by huckl3b3rry87

huckl3b3rry87, 2020-06-19:
This just happened again, a fix seems to be deleting the original default player start and adding a new one