mitchellspryn/UrdfSim #21: Setup PythonClient as a project in VS for using Python API

huckl3b3rry87, 2020-05-15:
I followed the tutorial on [setting up AIRsim](https://www.bing.com/videos/search?q=airsim+setup&docid=608005242406309788&mid=95959BAB60523B073D0295959BAB60523B073D02&view=detail&FORM=VIRE) for URDFSim and I am able to spawn the URDF
[image]

Now I am trying to control it, using the documentation[ here](https://microsoft.github.io/AirSim/apis/) and [here](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/UrdfClient.md), but alas, I am having issues. 

I think that it is due to my need to make the PythonClient into a project that is visible by my main project. For instance looking at URDFsim in VS, there are 13 projects

[image]

But, when I look at my project, again setup using the instructions in the tutorial video, I only have 3 projects and the PythonClient is missing. 
[image]

If you look at the folder structure of this, you can see that I copied the PythonClient folder from URDFsim into my project
[image]

Is there way to make it recognize PythonClient here so that it as a startup project?

Thank you!

mitchellspryn, 2020-05-15:
For the sake of your sanity, I would recommend having two different VS solutions - one for the python client, one for UrdfSim. Start UrdfSim in one VS instance. Click play in unreal. Then, start your python solution. It should connect.

Be sure to give the window inside Unreal focus. Otherwise, it will run really slowly.

huckl3b3rry87, 2020-05-16:
Sorry for the delay, I had to get Anaconda setup with VS. 

**Q1**: is there an easy way to do this:
> Be sure to give the window inside Unreal focus. Otherwise, it will run really slowly.

Unfortunately, I am still running into an issue.  Here are my steps:

1) open my main project with an instance of VS
2) right click on the .uproject and open with Unreal 
[image]
**Q1:** Is this needed or can I open the project from the File Explorer?

3) open a second instance of VS which is the PythonClient project 
4) right click on ArmClientTest.py and select ``Set as Startup File``
5) right click on PythonClient and select ``Set as Startup Project"
6) click play in Unreal 
8) click on the PythonClient solution 
[image]
7) click the green play button 

Then I get an error as 
[image]
```

  Message=__init__() got an unexpected keyword argument 'encoding'
  Source=C:\users\admin\documents\workspace\udds_urdf\b\pythonclient\msgpack\_packer.pyx
  StackTrace:
  File "C:\users\admin\documents\workspace\udds_urdf\b\pythonclient\msgpack\_packer.pyx", line 118, in msgpack._cmsgpack.Packer.__init__
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClient.py", line 9, in __init__
    self.airsim_client.confirmConnection();
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClientTest.py", line 17, in main
    client = ArmClient.ArmClient()
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClientTest.py", line 45, in <module>
    main()

```

So there seems to be a connection issue.

Please let me know if I am doing anything wrong or not needed, thanks!

huckl3b3rry87, 2020-05-16:
also, my AirSim settings, located in ``~/Documents/AirSim/`` are copied [from here](https://github.com/mitchellspryn/UrdfSim/blob/master/Examples/UrdfBot/settings-arm.json), except for the URDF path, which I fixed.
 
Additionally, I added ``  "RpcEnabled": true, `` 

as per [this AirSim issue](https://github.com/microsoft/AirSim/issues/278)

mitchellspryn, 2020-05-16:
**Q1**: Yes. Click on the viewport. That will give focus to Unreal.

**Q2**: I think you can just double-click on the .uproj. 

That's not a connection error, it looks like something's wrong with your install of msgpack or python. What versions do you have installed? Can you expand the "Python Environments" tab?

huckl3b3rry87, 2020-05-16:
Thank you very much for the fast response!

[image]

[image]

I think that I am using Anaconda 5.2.0
```
msgpack-python                     0.5.6
msgpack-rpc-python                 0.4.1
multipledispatch                   0.5.0
```

here are some more details
```
Backend Qt5Agg is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\ptvsd_launcher.py", line 119, in <module>
    vspd.debug(filename, port_num, debug_id, debug_options, run_as)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\debugger.py", line 37, in debug
    run(address, filename, *args, **kwargs)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_local.py", line 64, in run_file
    run(argv, addr, **kwargs)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_local.py", line 125, in _run
    _pydevd.main()
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\pydevd.py", line 1752, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\pydevd.py", line 1099, in run
    return self._exec(is_module, entry_point_fn, module_name, file, globals, locals)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\pydevd.py", line 1106, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\_pydev_imps\_pydev_execfile.py", line 25, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClientTest.py", line 45, in <module>
    main()
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClientTest.py", line 17, in main
    client = ArmClient.ArmClient()
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClient.py", line 9, in __init__
    self.airsim_client.confirmConnection();
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 89, in send_message
    self.connect()
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 98, in connect
    socket = ClientSocket(stream, self, self._encodings)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 53, in __init__
    BaseSocket.__init__(self, stream, encodings)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 12, in __init__
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
  File "msgpack\_packer.pyx", line 118, in msgpack._cmsgpack.Packer.__init__
TypeError: __init__() got an unexpected keyword argument 'encoding'
Press any key to continue . . .
```

mitchellspryn, 2020-05-16:
I meant this one: 
[image]

See is [this workaround](https://github.com/explosion/spaCy/issues/2810#issuecomment-425414118) helps you.

huckl3b3rry87, 2020-05-16:
[image]

huckl3b3rry87, 2020-05-16:
I just limited the verion, as you suggested
```
PS C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64> pip install "msgpack-numpy<0.4.4.0"
Collecting msgpack-numpy<0.4.4.0
  Using cached https://files.pythonhosted.org/packages/ad/45/464be6da85b5ca893cfcbd5de3b31a6710f636ccb8521b17bd4110a08d94/msgpack_numpy-0.4.3.2-py2.py3-none-any.whl
Requirement already satisfied: msgpack>=0.3.0 in c:\program files (x86)\microsoft visual studio\shared\anaconda3_64\lib\site-packages (from msgpack-numpy<0.4.4.0) (1.0.0)
Requirement already satisfied: numpy>=1.9.0 in c:\program files (x86)\microsoft visual studio\shared\anaconda3_64\lib\site-packages (from msgpack-numpy<0.4.4.0) (1.14.3)
Installing collected packages: msgpack-numpy
Successfully installed msgpack-numpy-0.4.3.2
You are using pip version 10.0.1, however version 20.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```
but same issue

huckl3b3rry87, 2020-05-16:
similar issues
https://github.com/explosion/spaCy/issues/2868
https://github.com/explosion/spaCy/issues/2810

mitchellspryn, 2020-05-16:
Hmm. This is odd. I just tried it on my end using the versions of packages that you have installed, and everything works. 

Does the script run if you execute it outside of VS? Like from the anaconda command prompt?

huckl3b3rry87, 2020-05-16:
@mitchellspryn thank you so much for your help:)

If I run Anaconda in the PowerShell, I get the same issue"
```
PS C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot> python.exe .\ArmClientTest.py
Traceback (most recent call last):
  File ".\ArmClientTest.py", line 45, in <module>
    main()
  File ".\ArmClientTest.py", line 17, in main
    client = ArmClient.ArmClient()
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot\ArmClient.py", line 9, in __init__
    self.airsim_client.confirmConnection();
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 89, in send_message
    self.connect()
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 98, in connect
    socket = ClientSocket(stream, self, self._encodings)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 53, in __init__
    BaseSocket.__init__(self, stream, encodings)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 12, in __init__
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
  File "msgpack\_packer.pyx", line 118, in msgpack._cmsgpack.Packer.__init__
TypeError: __init__() got an unexpected keyword argument 'encoding'
PS C:\Users\admin\Documents\workspace\UDDS_URDF\B\PythonClient\urdfbot>
```

huckl3b3rry87, 2020-05-16:
for reference, here are my packages, I started changing to the some of the ones you have, but still no luck
```
PS C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64> pip list
Package                            Version
---------------------------------- ---------
alabaster                          0.7.10
anaconda-client                    1.6.14
anaconda-navigator                 1.8.7
anaconda-project                   0.8.2
asn1crypto                         0.24.0
astroid                            1.6.3
astropy                            3.0.2
attrs                              18.1.0
Babel                              2.5.3
backcall                           0.1.0
backports.shutil-get-terminal-size 1.0.0
beautifulsoup4                     4.6.0
bitarray                           0.8.1
bkcharts                           0.2
blaze                              0.11.3
bleach                             2.1.3
bokeh                              0.12.16
boto                               2.48.0
Bottleneck                         1.2.1
certifi                            2018.4.16
cffi                               1.11.5
chardet                            3.0.4
click                              6.7
cloudpickle                        0.5.3
clyent                             1.2.2
colorama                           0.3.9
comtypes                           1.1.4
conda                              4.5.4
conda-build                        3.10.5
conda-verify                       2.0.0
contextlib2                        0.5.5
cryptography                       2.2.2
cycler                             0.10.0
Cython                             0.28.2
cytoolz                            0.9.0.1
dask                               0.17.5
datashape                          0.5.4
decorator                          4.3.0
distributed                        1.21.8
docutils                           0.14
entrypoints                        0.2.3
et-xmlfile                         1.0.1
fastcache                          1.0.2
filelock                           3.0.4
Flask                              1.0.2
Flask-Cors                         3.0.4
gevent                             1.3.0
glob2                              0.6
greenlet                           0.4.13
h5py                               2.7.1
heapdict                           1.0.0
html5lib                           1.0.1
idna                               2.6
imageio                            2.3.0
imagesize                          1.0.0
importlib-metadata                 1.6.0
ipykernel                          4.8.2
ipython                            6.4.0
ipython-genutils                   0.2.0
ipywidgets                         7.2.1
isort                              4.3.4
itsdangerous                       0.24
jdcal                              1.4
jedi                               0.12.0
Jinja2                             2.10
[... 110 lines cut ...]
tblib                              1.3.2
terminado                          0.8.1
testpath                           0.3.1
toolz                              0.9.0
tornado                            4.5.3
traitlets                          4.3.2
typing                             3.6.4
unicodecsv                         0.14.1
urllib3                            1.22
wcwidth                            0.1.7
webencodings                       0.5.1
Werkzeug                           0.14.1
wheel                              0.31.1
widgetsnbextension                 3.2.1
win-inet-pton                      1.0.1
win-unicode-console                0.5
wincertstore                       0.2
wrapt                              1.10.11
xlrd                               1.1.0
XlsxWriter                         1.0.4
xlwings                            0.11.8
xlwt                               1.3.0
zict                               0.1.3
zipp                               3.1.0
``

mitchellspryn, 2020-05-16:
Oh. That's actually a good catch. That means that the environment that VS is using is different from the one you're inspecting. 

I notice that you have 2 anacondas installed. Could you be using the wrong one? What happens if you select the other one?

huckl3b3rry87, 2020-05-16:
First I installed anaconda directly on windows, then I realized I could add it to my VS build, which I did and that is the one I think that I am using

If I switch to the one on my machine 

[image]

Then I have to start installing some missing packages, which I did on the anaconda version that I added directly to my VS

huckl3b3rry87, 2020-05-16:
for reference, I had to install 
```
pip install msgpack
pip install msgpack-rpc-python
pip install keyboard
pip install jsonpickle
```

but I did not try it on the Anaconda that I first installed just with Windows

mitchellspryn, 2020-05-16:
I see. 

What happens if you run this code? Do you get a traceback?
```python
import msgpack
p = msgpack.Packer(encoding='UTF-8', default=lambda x: x.to_msgpack())
```

huckl3b3rry87, 2020-05-16:
```
Anaconda 5.2.0 interactive window [PTVS 15.9.18254.1-15.0]
Type $help for a list of commands.
>>> import msgpack
>>> p = msgpack.Packer(encoding='UTF-8', default=lambda x: x.to_msgpack())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "msgpack\_packer.pyx", line 118, in msgpack._cmsgpack.Packer.__init__
TypeError: __init__() got an unexpected keyword argument 'encoding' 
```

mitchellspryn, 2020-05-16:
Ok, great. It runs fine for me. Progress. 

What is in C:\ProgramData\Anaconda3\Lib\site-packages\msgpack\_version.py? I have this:
```python
version = (0, 5, 6)
```

mitchellspryn, 2020-05-16:
I think I see the problem. Can you uninstall msgpack, msgpack-python, msgpack-rpc-python, and msgpack-numpy. Then, reinstall msgpack-python and msgpack-rpc-python, leaving out msgpack and msgpack-numpy?

huckl3b3rry87, 2020-05-16:
I should have mentioned that, yeah I had to add ``msgpack`` to install ``msgpack-python`` I just did what you said and started to add the packages and got
```
PS C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64> pip install msgpack-python
Collecting msgpack-python
  Using cached https://files.pythonhosted.org/packages/8a/20/6eca772d1a5830336f84aca1d8198e5a3f4715cd1c7fc36d3cc7f7185091/msgpack-python-0.5.6.tar.gz
Building wheels for collected packages: msgpack-python
  Running setup.py bdist_wheel for msgpack-python ... done
  Stored in directory: C:\Users\admin\AppData\Local\pip\Cache\wheels\d5\de\86\7fa56fda12511be47ea0808f3502bc879df4e63ab168ec0406
Successfully built msgpack-python
distributed 1.21.8 requires msgpack, which is not installed.
Installing collected packages: msgpack-python
Successfully installed msgpack-python-0.5.6
You are using pip version 10.0.1, however version 20.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

mitchellspryn, 2020-05-16:
Odd. What happens if you just remove msgpack?

huckl3b3rry87, 2020-05-16:
I guess it installed, it just gave this red issue
```
distributed 1.21.8 requires msgpack, which is not installed.
```
so, last time, I installed ``msgpack``

Currently I don't have ``msgpack``. Maybe it will work?

huckl3b3rry87, 2020-05-16:
And!! The URDF robot is moving around amazingly:)

Thank you so much @mitchellspryn !

closed 2020-05-16 by huckl3b3rry87

mitchellspryn, 2020-05-16:
Great. Glad we could get it up and running.