mitchellspryn/UrdfSim #22: getting the robot states

huckl3b3rry87, 2020-05-18:
I am interested in getting [the robot states](http://wiki.ros.org/urdf/Tutorials/Using%20urdf%20with%20robot_state_publisher).

So, for instance, in the case of the [Arm](https://github.com/mitchellspryn/UrdfSim/blob/master/docs/images/Arm.gif), these states include the joint angles and the position of the prismatic joint of the grippers.

For reference, I am interested in doing some closed loop, probably PID control to move the arm to a particular location and picking it up and moving it.

Thank you!

huckl3b3rry87, 2020-05-18:
Sorry, I think that I found it `` getUrdfBotState``.

closed 2020-05-18 by huckl3b3rry87

mitchellspryn, 2020-05-18:
For anyone coming from google, yes, that is the API call to use.