mitchellspryn/UrdfSim #23: motion planning and control backend

huckl3b3rry87, 2020-05-18:
Correct me if I am wrong, but it seems that there is no API or backend available to do motion planning and control, for instance in ROS, there is [moveit](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/urdf_srdf/urdf_srdf_tutorial.html). 

I am interested in setting the position of the gripping, in the Arm URDF, so that I can pick up an object and move it around. To accomplish this, I will need to do trajectory/motion planning (maybe just inverse kinematics) and then trajectory following (low level control like PID) for each joint. 

While I will enjoy writing this kind of code, time is a constraint for me and I was wondering if you had any ideas for backends that we may be able to establish for doing such a tasks.

mitchellspryn, 2020-05-18:
There is nothing like that built into UrdfSim. 

MoveIt is good. Writing a ros node to publish data from your bot into ROS should be pretty easy. 

[The Open Motion Planning Library](https://ompl.kavrakilab.org/) might be of interest to you as well.

huckl3b3rry87, 2020-05-19:
@mitchellspryn thank you very much for the advice!

closed 2020-05-19 by huckl3b3rry87