mitchellspryn/UrdfSim #24: Issue controlling gripper in RobotArm.xml

huckl3b3rry87, 2020-05-20:
I am trying to control the manipulator to open or close using functionality from ArmClientTest.py i.e.,

```
print('Closing claw')
client.set_manipulator(0)
```
or 
```
print('Opening claw')
client.set_manipulator(1)
```

But it is not moving the manipulator arms.

I am able to use the PythonClient to move the URDF around to random pose.

**Q1**: Any thoughts? 

**Q2**: Eventually, I would like to be able to grab this cup (or some object) and move it around. Is this easily done once I can grab the cup? 

Thank you!

huckl3b3rry87, 2020-05-21:
Also, I went back to the default URDF for this robot ( I had modified a couple things), but I am still getting the issue
[image]

huckl3b3rry87, 2020-05-21:
It seems to be an issue with the prismatic joint, if I change 
```
  <joint name="manipulator_base_to_right_manipulator" type="prismatic">
    <parent link="manipulator_base"/>
    <child link="right_manipulator"/>
    <origin rpy="0 0 0" xyz="0 -0.085 0.065"/>
    <axis rpy="0 0 0" xyz="0 -1 0"/>
    <limit lower="-0.070" upper="0" effort="10000" velocity="0.2"/>
  </joint>
```
to 
```
  <joint name="manipulator_base_to_right_manipulator" type="revolute">
    <parent link="manipulator_base"/>
    <child link="right_manipulator"/>
    <origin rpy="0 0 0" xyz="0 -0.085 0.065"/>
    <axis rpy="0 0 0" xyz="1 0 0"/>
    <limit lower="-1.57079" upper="1.57079" effort="10000" velocity="0.2"/>
  </joint>
```
then, when the simulation starts, it looks like this 
[image]

then sending
```
 client.set_manipulator(0)
```
rotates the right claw as 

[image]

huckl3b3rry87, 2020-05-21:
Q2: Eventually, I would like to be able to grab this cup (or some object) and move it around. Is this easily done once I can grab the cup? 

A2: yes

mitchellspryn, 2020-05-29:
This is truely bizarre. I tried it on my box just now. Everything seems to work as I would expect it to. A thought - did you rotate the playerstart from 0 degrees?

huckl3b3rry87, 2020-05-29:
Just started from a fresh install and the gripper is closing, I did not test the rotate playerstart in this new version. 

If I have future issues, I will comment here

Thanks!

closed 2020-05-29 by huckl3b3rry87