mitchellspryn/UrdfSim #26: Using a static camera with URDFSim

huckl3b3rry87, 2020-05-29:
I am trying to make it so that the camera that is used is static. I followed this tutorial
https://www.youtube.com/watch?v=7dZoISdc0dM

i.e.,
[image]

But, similar to [this issue](https://github.com/microsoft/AirSim/issues/2452), when I change it to AirSimMode, which I need to spawn the URDF bot, then it reverts back to the camera on the URDFBot.

Is there a way that I can spawn the Urdfbot without having to use AirSimMode? Or is there another way to deal with this?

Thank you very much

mitchellspryn, 2020-05-29:
I have no idea what your blueprint is trying to do. It's probably not relevant, as we don't use blueprints for controlling the camera. 

To concretely answer your questions:

**Q1**: Realistically, no. Even if you could, you wouldn't have any of the infrastructure to grab sensor information, the rpc server to communicate with it, etc.

**Q2**: Yes. 

Looking in [CameraDirector.cpp](https://github.com/mitchellspryn/UrdfSim/blob/master/Unreal/Plugins/AirSim/Source/CameraDirector.cpp), there appears to be a CAMERA_MODE_MANUAL. Looks like it's bound to the "M" key. Maybe that would suit your needs?

If that doesn't, as a quick hack you could try is commenting out the attachSpringArm() calls in ACameraDirector::setMode(). That should stop it from being attached to the bot, and leaving it stuck in place.

huckl3b3rry87, 2020-05-29:
@mitchellspryn thank you again for you helpful advice.

For your first suggestion:

I actually tried to use a BP_CameraDirector after watching [this AirSim tutorial](https://www.youtube.com/watch?v=1oY8Qu5maQQ), but I ran into a few issues 
1: The UrdfBotPawn is not a blueprint, so I could not just drag it into the unreal editor
2: Even if I could drag it in, I am missing the Pawn field in BP_CameraDirector

On the video
[image]

Mine
[image]

Since I do not have this setup, I assume why when I hit the "M" key, nothing happens. 

**Q3** Do I need to have this camera director setup to use the "M" mode?

**Q4** If **Q3** is yes, do you know what I may be doing wrong?

**Q5** if **Q3** is not, is there some other thing that I need to do to enable "M" mode?

Thanks!

mitchellspryn, 2020-05-29:
You shouldn't have to use blueprints / configure anything. It should all happen through your settings.json. I'm not sure why pressing the M key isn't working for you - the keybindings might be broken. 

What happens if you set "ViewMode" for your camera to "Manual" (from [here](https://github.com/mitchellspryn/UrdfSim/blob/master/AirLib/include/common/AirSimSettings.hpp))?

huckl3b3rry87, 2020-05-29:
OK. 

Maybe [my settings.json](https://gist.github.com/huckl3b3rry87/5f7361410150a908395f1d10f3dcdd08) is messed up? Do I need to add anything else from [here](https://microsoft.github.io/AirSim/settings/#available-settings-and-their-defaults)?

If I change [this line ](https://github.com/mitchellspryn/UrdfSim/blob/master/AirLib/include/common/AirSimSettings.hpp#L534) to 
```
std::string view_mode_string = "Manual";
```
and recompile the project in the unreal editor, I am still unable to move the camera using the keyboard

huckl3b3rry87, 2020-05-30:
> If that doesn't, as a quick hack you could try is commenting out the attachSpringArm() calls in ACameraDirector::setMode(). That should stop it from being attached to the bot, and leaving it stuck in place.

This did not work, the camera is still attached to the bot

mitchellspryn, 2020-05-31:
Wait. I'm confused. You're using the arm bot, right? Can't you just attach the camera to the base, which shouldn't move at all?

mitchellspryn, 2020-06-01:
Just checked the code. Looks like SimModeUrdfBot doesn't use CameraDirector. So, there isn't an easy way to do this via configuration.

As a hack, if you don't also need moving cameras, you can comment out line 154 of [PawnSimApi.cpp](https://github.com/mitchellspryn/UrdfSim/blob/master/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp) (the camera->AttachToComponent()). This should prevent the camera from moving once spawned.

huckl3b3rry87, 2020-06-01:
@mitchellspryn thank you very much, your suggestion and modifying the settings.json file to move the camera worked~
```
        "Arm": {

          "X": -80,

          "Y": 0,

          "Z": -20,

          "Pitch": 0,

          "Roll": 180,

          "Yaw": 0,

          "AttachLink": "manipulator_base"

        }
```

closed 2020-06-01 by huckl3b3rry87