mitchellspryn/UrdfSim #27: collisions not working as expected

huckl3b3rry87, 2020-06-04:
I am trying to use the robot arm to pick up objects, but it seems that the gripper finger is able to go straight through an object, especially when I decrease the mass of the object.

Consider this sequence example
[image]

[image]

[image]

** the mustard bottle passes straight through the manipulator hand 

I have complex collisions set for this object
[image]

and, for the manipulator arms, I have added collisions the same as the visual, e.g.,
```
  <link name="left_manipulator">
    <visual>
      <geometry>
        <box size="0.06 0.035 0.15"/>
      </geometry>
      <material name="rubber"/>
    </visual>
    <collision>
      <geometry>
        <box size="0.06 0.035 0.15"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="10"/>
      <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
    </inertial>
  </link>
```
Might the behavior be expected using the physics engine that Unreal uses? 

If so, is there any way to increase the fidelity of the simulation to avoid this?

Or if not, any idea what else may be going on here.

Thanks :)

mitchellspryn, 2020-06-04:
[You're going to want to read this](https://docs.unrealengine.com/en-US/Engine/Physics/SimpleVsComplex/index.html).

I assume you want to move the bottle. In that case, you will need to enable physics (there should be a checkbox hidden somewhere in the options). When physics is enabled, you can only use simple collision, which restricts you to convex hulls (although you can have multiple). This is a common restriction for most game engines that I'm aware of, and it's done for performance reasons. 

So, I'd recommend ditching the complex collision, adding simple collision (via the 26 DOP option), and enabling physics.

huckl3b3rry87, 2020-06-04:
@mitchellspryn I did mess with both Simple and Complex collisions, as well as enabling physics, but I did not try Only Simple and enable physics, as you suggested, which works!

Thanks!

huckl3b3rry87, 2020-06-04:
OK, I think that I got it, if I reduce the Max Physics Delta time it works as I would like 

[image]

@mitchellspryn thanks again:)

closed 2020-06-04 by huckl3b3rry87

mitchellspryn, 2020-06-04:
You're also going to want to enable substepping, and substepping async if your GPU supports it (it probably does).

huckl3b3rry87, 2020-06-04:
@mitchellspryn thank you for the suggestions!

Yeah, I ended up turned the coefficient of friction to 10,000.... I just want it picked up