mitchellspryn/UrdfSim #28: Floating Lunabot

mtmal, 2020-06-04:
Hello,

First of all, thanks for this nice simulator. I'm looking for an inspiration to implement custom Mars-like rover into UE, but the game engine itself supports only cars, where there is a single engine to power the whole vehicle. I learnt about AirSim and UrdfSim only earlier this week and I am trying to evaluate whether it can meet my requirements. 

I managed to successfully build UrdfSim on Ubuntu and run few examples and tests. I finally launched Lunabot but it seems to be floating above the ground (please see the picture below).

[image]

When using LunabotClientTest.py commands are correctly sent (wheels/arm are turning) and sensors data is correctly received. I imagine that the robot should be 'dropped' at the start of the simulation when physics kicks in. For comparison, I also checked the following:
- RobotArm's base is also floating above the ground
- HelloCar example works correctly - both C++ and Python examples

Thanks,
Mateusz

huckl3b3rry87, 2020-06-04:
this is strange, when I run RobotArm base, it drops to the ground. 

It may be worth trying to start from scratch with the minimal changes that you need to run on Linux - you may have accidentally messed something up, which happened to me.

mitchellspryn, 2020-06-04:
If you run pxvis collision, can you see the collision being generated?

mtmal, 2020-06-05:
@huckl3b3rry87 I rebuilt Blocks environment and AirSim plugin without any luck. Besides, I didn't make any changes just yet, to get it all working. The only think that comes to my mind is that I had to rebuild the editor with bUseRTTI set to true, as otherwise it was preventing dynamic casting on the AirSim/UrdfSim side.

@mitchellspryn it seems that collisions are generated. I tried to play with various collision settings for capsule component, but without any luck.

[image]
[image]

I know it might be hard to spot these magenta lines, especially on arm base, but they are there, just z-fighting with the base. I will try to look more into these collisions, although I would very much appreciate any more hints. Thanks!

mtmal, 2020-06-05:
Ok, the problem is solved. When inspecting UrdfBotPawn I noticed that its static mesh did not have "Simulate Physics" set:

[image]

This was a bit odd, as I looked into the code and wherever static mesh is created, there is a call to SetSimulatePhysics(true). However, I decided to add
`this->root_component_->GetRootMesh()->SetSimulatePhysics(true);`
into the UrdfBotPawn::ConstructFromFile which solved the problem.

I can prepare a PR with my fixes for Linux build if you would like, but I do not have Windows nor Mac to test these changes.

seshanrv, 2020-06-12:
@mtmal  I am trying to use UrdfSim in Linux for simulating an agricultural tractor which has an implement, it is somewhat similar to your application. I am facing similar errors as discussed in #15. Could you share your fixes?

mtmal, 2020-06-13:
Hi @seshanrv,
I created a pull request with updates to compile this project on Linux with UE 4.24.3. Should work also with 4.25 version.
Cheers,
Mateusz

seshanrv, 2020-06-15:
@mtmal  Thank you for the PR. I tried your version of the plugin in Ubuntu 18.04 LTS, the earlier errors are gone but I face new issues:
1. I don't see a .uproject file for the Blocks environment, after I build the plugin. (This is true for the current master branch as well).
2. When I try to add it to a different project I get an error in the linking step, as seen below. In fact the same error is repeated for - `typeinfo for ACharacter', ....AGameModeBase', etc. 

Do you have a suggestion to fix these? 

Building UE4Editor...
Performing 2 actions (10 in parallel)
[1/2] Link (ld) libUE4Editor-AirSim-Linux-Debug.so
/home/seshan/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: /home/seshan/Documents/Unreal Projects/MyProject/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Debug/AirSim/Module.AirSim.cpp.o: in function `_ZN16UE4Casts_Private11DynamicCastIP13AirsimVehicle6AActorEEN9TEnableIfIXntsr4TAndI10TIsPointerIT_E4TAndIJ20TIsCastableToPointerIN14TRemovePointerIS6_E4TypeEE11TIsCastableIT0_EEEEE5ValueES6_E4TypeEPSF_':
/home/seshan/UnrealEngine/Engine/Source/Runtime/CoreUObject/Public/Templates/Casts.h:(.text._ZN16UE4Casts_Private11DynamicCastIP13AirsimVehicle6AActorEEN9TEnableIfIXntsr4TAndI10TIsPointerIT_E4TAndIJ20TIsCastableToPointerIN14TRemovePointerIS6_E4TypeEE11TIsCastableIT0_EEEEE5ValueES6_E4TypeEPSF_[_ZN16UE4Casts_Private11DynamicCastIP13AirsimVehicle6AActorEEN9TEnableIfIXntsr4TAndI10TIsPointerIT_E4TAndIJ20TIsCastableToPointerIN14TRemovePointerIS6_E4TypeEE11TIsCastableIT0_EEEEE5ValueES6_E4TypeEPSF_]+0x21): undefined reference to `typeinfo for AActor'

mtmal, 2020-06-15:
Hi @seshanrv,
1) As .uproject is listed in .gitignore, no files are pushed, however, you can use one below:
`{
	"FileVersion": 3,
	"EngineAssociation": "{1D709676-2A05-101A-000D-0091F40A9A2B}",
	"Category": "",
	"Description": "",
	"Modules": [
		{
			"Name": "Blocks",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
				"AirSim"
			]
		}
	],
	"Plugins": [
		{
			"Name": "AirSim",
			"Enabled": true
		},
		{
			"Name": "SteamVR",
			"Enabled": false
		},
		{
			"Name": "OculusVR",
			"Enabled": false
		}
	]
}`

2) this is issue is related to RTTI. Not sure if that's because AirSim/UrdfSim use STL, I'm not that proficient in C++ (thought I would love to be). What I have done, was to edit Engine/Source/Programs/UnrealBuildTool/System/CppCompileEnvironment.cs and Engine/Source/Programs/UnrealBuildTool/Configuration/ModuleRules.cs by setting bUseRTTI to true, and forcing " -frtti" in Engine/Source/Programs/UnrealBuildTool/Platform/Linux/LinuxToolChain.cs (inside GetRTTIFlag function). You would then need to rebuild Unreal Engine.

Mateusz

seshanrv, 2020-06-22:
@mtmal thank you! I have it working now.