mitchellspryn/UrdfSim #34: How Can I make movement of my RunaBot?

guldamkwak3114, 2021-02-15:
I manaed to import my Rubabot to the Three Bridges environment but have no idea how to let them move. How Can I let them move? 

[image]

mitchellspryn, 2021-02-15:
You'll need to use the API, either via C++ or python. [Here are some examples](https://github.com/mitchellspryn/UrdfSim/tree/master/PythonClient/urdfbot). [Here is the one relevant to the lunabot](https://github.com/mitchellspryn/UrdfSim/blob/master/PythonClient/urdfbot/LunabotClientTest.py).

guldamkwak3114, 2021-02-16:
Can I use xbox controller, like in the case of quadrotor in Airism?

mitchellspryn, 2021-02-16:
Sure, but you'll have to write the code to map button presses to API actions. 

URDFSim cannot provide this functionality "out-of-the-box", because we don't know a-priori what your robot looks like.

closed 2021-03-05 by mitchellspryn