mitchellspryn/UrdfSim #36: Can I pinpoint the location of cause of my errors ?

guldamkwak3114, 2021-02-17:
Trying to lendering my URDF file, I constantly come across error message, like

"error at startup : Error parsing 3-d vector from string , did not detect 3 elements" 

Is there any method that I can pin-point the location of cause of errors in my URDF file ?

mitchellspryn, 2021-02-17:
Searching through the code yields [this](https://github.com/mitchellspryn/UrdfSim/blob/a78e6301220fbc97724fd742ceb5ec04270f638d/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfParser/UrdfParser.cpp#L1132) or [this](https://github.com/mitchellspryn/UrdfSim/blob/a78e6301220fbc97724fd742ceb5ec04270f638d/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/MeshGeneration/StaticMeshGenerator.cpp#L281) as the likely sources of error. Likely, the syntax for your URDF file isn't correct. In particular, make sure that the syntax for specifying all vectors looks [like this](https://github.com/mitchellspryn/UrdfSim/blob/master/Examples/UrdfBot/LunabotFromMesh.xml#L48). 

If that still doesn't solve, then you can try placing a debug breakpoint on those lines and examine the stack trace. It should become apparent which part of the URDF is giving the program problems.

guldamkwak3114, 2021-02-18:
How Can I set debug breakpoint and check the error message and stack trace ??

mitchellspryn, 2021-02-18:
[This documentation page might be helpful](https://docs.microsoft.com/en-us/visualstudio/debugger/using-breakpoints?view=vs-2019).

guldamkwak3114, 2021-02-18:
I mean, I am trying to render my custom robot using my URDF in RMSpringLandscape. When I open RMSpringLandscape.exe file with correct settings.json with correct urdf/mesh file path, it shows a number of different erors. If I am dealing with cpp source code file directly I can use the debugger and check the stack trace. Howerver, if the compiled URDFSim file raises erros when I try to render my URDF file, where Can I check the stack trace?

mitchellspryn, 2021-02-18:
Oh. Then this is on the wrong repo :). 

There isn't a great way once the application has been compiled. Perhaps post your URDF, and I can take a look to see if I can spot anything obvious.

guldamkwak3114, 2021-02-19:
I solved my issues. Thanks. 

1. If you use complex mesh file for your URDFbot, URDFSim fail to render it even if it can be rendered in Gazebo. Try mesh files which is as simple as possible 

2. Be careful about the path of the urdf file and mesh file 

3. When you set the coordinates

      <origin
        rpy="-0.0118705871576861 0.0126211689490765 0.428852289941002" xyz="0 0 0" />
        
        the rpy should come first.

mitchellspryn, 2021-02-19:
Interesting. Thanks for digging into this.

(1) is known. It turns out that PhysX will crash if you have too many linkages together. If you look at the lunabot, you'll see that while it could be represented by a bunch of boxes, it's been fused together into a mesh for this exact reason. 

(3) sounds like a bug. I'm suprised that would crash the system.

closed 2021-03-05 by mitchellspryn