mitchellspryn/UrdfSim #37: UrdfBot error

henry-tra, 2021-02-18:
Hi, I wanted to start with the provided Examples but I am not able to insert the Lunabot.

I am pretty new to the Unreal Engine editor so I tried the following:

I have changed the path to the xml in the json file but I did not change anything else in the provided json file. Afterwards I have put the json file into the /Documents/Airsim folder. But when I press play the Unreal editor says "SimMode is not valid: UrdfBot" and "There were no compatible vehicles created for current SimMode! Check your settings.json.". 

Is there something missing or do I have to add something else?

mitchellspryn, 2021-02-18:
IIRC you need a settings.json, which looks like this and would go in your documents folder. You would then need a URDF which looks like [this](https://github.com/mitchellspryn/UrdfSim/blob/master/Examples/UrdfBot/LunabotFromMesh.xml), and change the JSON to point to it. Sounds like you already tried that, so the next step is to try to step through the code and see where the problem is. The errors look like they're coming from [here](https://github.com/mitchellspryn/UrdfSim/blob/a78e6301220fbc97724fd742ceb5ec04270f638d/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L194) and [here](https://github.com/mitchellspryn/UrdfSim/blob/a78e6301220fbc97724fd742ceb5ec04270f638d/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L314). Perhaps try setting breakpoints there might illuminate the problem.

henry-tra, 2021-02-18:
I understand what you mean with setting breakpoints but I do not know how set ones since I have just started the Unreal Engine Editor and just pressed play and not within some kind of IDE like Visual Studio: 
[image]

Can you explain this more in detail please

mitchellspryn, 2021-02-18:
For UrdfSim, the only supported platform is windows. This is the easiest way to edit the c++ files, as it generates visual studio solutions. 

If you are able to build in linux, you can try attaching GDB to the exe as you run it.

guldamkwak3114, 2021-02-19:
You need to run an environment that had been compiled for URDFSim, not Airsim. Threebridges 

https://github.com/mitchellspryn/AnnotatedUnrealMaps

is one example of such enrionment.

closed 2021-03-30 by henry-tra