mitchellspryn/UrdfSim #38: Can I compile custom environment using URDFSIm?

guldamkwak3114, 2021-02-19:
I succeded in rendering my own URDFbot, However, I am having struggle in compiling my own enviornment to run my URDFBot

If I want to compile my own environment for URDFSIm, Can I just follow 

https://microsoft.github.io/AirSim/unreal_custenv/

This procedure, with only minor modification with copying Unreal/Airsim folder form URDFSIm foldre, not AirsimFoler? 
Does this minor modification works?

mitchellspryn, 2021-02-19:
Yup, that's how I generated the AnnotatedUnrealMap. I copied the folder into the `Plugins` directory, and made my uproject file look like this:

```
{
	"FileVersion": 3,
	"EngineAssociation": "4.24",
	"Category": "",
	"Description": "",
	"Modules": [
		{
			"Name": "RMSpringLandscape",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
				"AirSim"
			]
		}
	],
	"Plugins": [
		{
			"Name": "AirSim",
			"Enabled": true
		},
		{
			"Name": "PythonScriptPlugin",
			"Enabled": true
		},
		{
			"Name": "EditorScriptingUtilities",
			"Enabled": true
		},
		{
			"Name": "SequencerScripting",
			"Enabled": true
		}
	]
}
```

You might not need the last three plugins, I did some python scripting to automate building some of the environment.

guldamkwak3114, 2021-02-19:
Could you try if you are able to compile LandscapeMountains using URDFSim? I keep coming acrossing lots of errors in trying them.

mitchellspryn, 2021-02-23:
I could, but that's probably not the best way to handle this issue, as dropping random binaries in github releases doesn't scale well. Better if we can fix the compile errors. What sorts of errors are you seeing? Are you building with VS 2017 on UE 4.24?

guldamkwak3114, 2021-03-03:
Those are some of errors I got when I try compile City Park Environment
https://www.unrealengine.com/marketplace/en-US/product/city-park-environment-collection?sessionInvalidated=true

[image]

which shows no problem when I try compile them using airsim. Could you try a look?

mitchellspryn, 2021-03-04:
Can you post the full build output? Screenshots don't show the full error message.

mitchellspryn, 2021-03-04:
It looks like you've opened a duplicate issue [here](https://github.com/mitchellspryn/UrdfSim/issues/41), so closing this one.

closed 2021-03-04 by mitchellspryn