mitchellspryn/UrdfSim #39: Using custom mesh with texture

hanmmmmm, 2021-02-26:
Hi! How can I use my custom mesh with texture in urdfsim?

The _<mesh >_ tag in "settings.json" needs the 'type' and 'location' parameters. 
I tried to put an obj mesh file in the "location", with stl_ascii type, then unreal says this file has no endsolid when clicking the play button. Indeed, an obj file does not have 'endsolid'. 

So could I use an obj/dae mesh file by just modifying "settings.json"?   
if not, could I make these files into uasset files, and import them as _<mesh>_ in "settings.json"? 

Thank you!

hanmmmmm, 2021-02-26:
Okay, I tried to make **fbx** mesh file with texture, and import them into unreal as 'static mesh'. Like the picture.

[image]

Then make change in the urdf .XML file :
```
<mesh type="unreal_mesh" location="StaticMesh'/Game/mesh/box.box'" reverse_normals="false" scale_factor="0.1"/>
<!--material name="rubber"/-->
```

Now the robot has my custom mesh and custom texture. But I still need to figure out the link pose coordinates and collision because the robot flies randomly when starting play. 

[image]

mitchellspryn, 2021-02-26:
Regarding your first question - obj mesh is not supported for reading from a text file, only stl ascii. 

If you are comfortable with the unreal workflow, importing the mesh into unreal and creating a .uasset is definitely the way to go. You'll get better lighting (it can be pre-baked), as well as the ability to better optimize the geometry with the included tools (e.g. PHAT). 

It looks like you've just swapped out your mesh with the central mesh of the lunabot? The physics constraints are probably going crazy, which is why the bot is falling over. I'd try removing all of the other elements in the URDF, and only importing your box. Then, if you have more meshes, build it up slowly piece by piece.

hanmmmmm, 2021-02-26:
Thank you for replying!   Yeah, I replaced a part of lunabot with my simple box. So the physics would be not working properly. I will try to build a robot from scratch.  

My other question is what does the **CollisionBlacklist** do, for example, this part in the lunabot XML file:

`"CollisionBlacklist": [
                {
                    "BotMesh": "front_right_wheel", 
                    "ExternalActorRegex": "Ground"
                },
                {
                    "BotMesh": "back_right_wheel", 
                    "ExternalActorRegex": "Ground"
                },
                {
                    "BotMesh": "front_left_wheel", 
                    "ExternalActorRegex": "Ground"
                },
                {
                    "BotMesh": "back_left_wheel", 
                    "ExternalActorRegex": "Ground"
                }
            ],`

mitchellspryn, 2021-02-26:
The framework will generate a collision event whenever one of your meshes collides with another one (that is not connected via a physics constraint). However, considering that your robot is on the ground, you will likely be continuously generating collision events between the bottom of your bot and the ground. In addition to being annoying, this is a performance issue. 

This blacklist allows you to filter out collision events. Here, it's saying to ignore collisions between the wheels and each member matching "Ground".