mitchellspryn/UrdfSim #40: Error at Building

guldamkwak3114, 2021-03-04:
I installed VS2017 and window SDK 8.1 when I try to build URDFSim

(1) I encounter the error that there is no Eigen/Dense folder in URDFSim Repository. I manually install Eigen and copy the contents of Eigen library to UrdfSim/Airlib/include folder so that looks like below. 

[image]

I hope you check this issue

(2) Besides the error in (1), I encouter erros like

LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\DroneShell\DroneShell.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\HelloDrone\HelloDrone.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\DroneServer\DroneServer.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\Examples\Examples.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\AirLibUnitTests\AirLibUnitTests.vcxproj]
LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\HelloCar\HelloCar.vcxproj]
Done Building Project "C:\UrdfSim\HelloCar\HelloCar.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\UrdfSim\AirSim.sln" (default targets) -- FAILED

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
(Link target) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\DroneShell\DroneShell.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\HelloDrone\HelloDrone.vcxproj" (default target) (5) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\HelloDrone\HelloDrone.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\DroneServer\DroneServer.vcxproj" (default target) (6) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\DroneServer\DroneServer.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj" (default target) (7) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\Examples\Examples.vcxproj" (default target) (8) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\Examples\Examples.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\AirLibUnitTests\AirLibUnitTests.vcxproj" (default target) (11) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\AirLibUnitTests\AirLibUnitTests.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\HelloCar\HelloCar.vcxproj" (default target) (12) ->
  LINK : fatal error LNK1104: cannot open file 'ucrtd.lib' [C:\UrdfSim\HelloCar\HelloCar.vcxproj]

I found that ucrtd.lib and urct.lib files are in window SDK folder. If I manually copy them into the above-mentioned folders, I found that those errors gone vanishing. However, I have no idea on why those errors occurs. 

Could you check? Thanks.

mitchellspryn, 2021-03-04:
Doesn't Eigen get downloaded when you run `Build.cmd`?

Why are you linking against windows 8.1 SDK? Shouldn't it be the windows 10 SDK? In any case, this sounds like your include path inside VS was set improperly - if you go to <project>->properties->VC++ directories, you should see $(WindowsSDK_LibaryPath_x64). 

[image]

guldamkwak3114, 2021-03-04:
If i use VS2019 with window sdk 10, when I runthen it shows

========================================

-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/UrdfSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeOutput.log".
See also "C:/UrdfSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeError.log".
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj

#### Build failed - see messages above.

========================================
If i use VS2017 with window sdk 10, when I run .\build.cmd, then it shows

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"C:\UrdfSim\AirLib\AirLib.vcxproj" (default target) (3) ->
"C:\UrdfSim\MavLinkCom\MavLinkCom.vcxproj" (default target) (4) ->
(_CheckWindowsSDKInstalled target) ->
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.WindowsSDK.targe
ts(46,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or chan
ge the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". [
C:\UrdfSim\MavLinkCom\MavLinkCom.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\Examples\Examples.vcxproj" (default target) (8) ->
"C:\UrdfSim\SGM\src\sgmstereo\sgmstereo_vc15.vcxproj" (default target) (9) ->
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.WindowsSDK.targe
ts(46,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or chan
ge the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". [
C:\UrdfSim\SGM\src\sgmstereo\sgmstereo_vc15.vcxproj]

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\Examples\Examples.vcxproj" (default target) (8) ->
"C:\UrdfSim\SGM\src\stereoPipeline\stereoPipeline_vc15.vcxproj" (default target) (10) ->
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.WindowsSDK.targe
ts(46,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or chan
ge the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". [
C:\UrdfSim\SGM\src\stereoPipeline\stereoPipeline_vc15.vcxproj]

================ 

The only configuration works is VS2017 along with SDK 8.1 

I tested windows 10.0.19041.685 

Could you configure what's wrong?

guldamkwak3114, 2021-03-04:
It seems VS2017 along with both of Win SDK 8.1 and SDK 10.0.19041.685 is the only configuration that works.
It does not automatically download eigen.

mitchellspryn, 2021-03-04:
I just tried a clean build, and [this part](https://github.com/mitchellspryn/UrdfSim/blob/master/build.cmd#L145) of the build script downloaded eigen. 

Good to know that it depends on 8.1  as well.

closed 2021-03-04 by mitchellspryn