mitchellspryn/UrdfSim #41: Errors at Custom Environment Build

guldamkwak3114, 2021-03-04:
I downloaded an Unreal Environment. Make C++ Class. Copy Plugins. Generate VIsual Studio Project FIles. Then run build from the sln file. However, I got 

[image]

Anybody has the ideas on how to debug those erros?
Thanks

mitchellspryn, 2021-03-04:
Please don't paste screenshots of errors, they don't contain the full stack trace. Please share the full build output.

These don't appear to come from inside URDFSim. Can you build wihtout urdfsim? I noticed that in your steps, you didn't include "edit the .uproject file to enable the plugin." Perhaps ask on the UE4 forums.

guldamkwak3114, 2021-03-04:
I edited the .uporject file so that it look like

{
	"FileVersion": 3,
	"EngineAssociation": "4.24",
	"Category": "",
	"Description": "",
	"Modules": [
		{
			"Name": "CityPark",
			"Type": "Runtime",
			"LoadingPhase": "Default",
            "AdditionalDependencies": [
                "AirSim"
            ]
		}
	],
    "TargetPlatforms": [
        "MacNoEditor",
        "WindowsNoEditor"
    ],
	"Plugins": [
		{
			"Name": "InstanceTool",
			"Enabled": false,
			"MarketplaceURL": "com.epicgames.launcher://ue/marketplace/content/0283702886e8467383899c7b791c4b40"
		},
        {
            "Name": "AirSim",
            "Enabled": true
        }
	]
}

==== 
when I run build the output sln file I face the error below

1>Building CityParkEditor...
1>Using Visual Studio 2017 14.16.27045 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023) and Windows 10.0.19041.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>Building 16 actions with 6 processes...
1>  [1/16] Module.AirSim.5_of_7.cpp
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/unsupported/test/cxx11_tensor_reduction.cpp(10): fatal error C1083: Cannot open include file: 'main.h': No such file or directory
1>  [2/16] tensor_benchmarks_sycl.cc
1>C:\CityPark\Plugins\AirSim\Source\AirLib\include\bench\tensors\tensor_benchmarks_sycl.cc(3): fatal error C1083: Cannot open include file: 'SYCL/sycl.hpp': No such file or directory
1>  [3/16] benchmark_main.cc
1>C:\CityPark\Plugins\AirSim\Source\AirLib\include\bench\tensors\benchmark_main.cc(17): fatal error C1083: Cannot open include file: 'regex.h': No such file or directory
1>  [4/16] Module.AirSim.2_of_7.cpp
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(1): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(1): error C2146: syntax error: missing ';' before identifier 'A'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(1): error C2825: 'MatrixXf': must be a class or namespace when followed by '::'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(1): error C2510: 'MatrixXf': left of '::' must be a class/struct/union
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(2): error C2143: syntax error: missing ';' before '<'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(2): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(3): error C3927: '->': trailing return type is not allowed after a non-function declarator
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(3): error C3484: syntax error: expected '->' before the return type
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(3): error C3613: missing return type after '->' ('int' assumed)
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(3): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(3): error C2146: syntax error: missing ';' before identifier 'compute'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(4): error C2143: syntax error: missing ';' before '<<'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(4): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(5): error C3927: '->': trailing return type is not allowed after a non-function declarator
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(5): error C3484: syntax error: expected '->' before the return type
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(5): error C3613: missing return type after '->' ('int' assumed)
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(5): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(5): error C2086: 'int schur': redefinition
1>  C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(3): note: see declaration of 'schur'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(5): error C2146: syntax error: missing ';' before identifier 'compute'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(6): error C2143: syntax error: missing ';' before '<<'
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(6): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:/CityPark/Plugins/AirSim/Source/AirLib/include/doc/snippets/RealSchur_compute.cpp(6): error C2086: 'int cout': redefinition
[... 486 lines cut ...]
1>C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\commctrl.h(7666): error C2056: illegal expression
[... cut ...]
1>c:\program files (x86)\windows kits\10\include\10.0.19041.0\um\shobjidl_core.h(23660): error C2061: syntax error: identifier 'MSG'
1>c:\program files (x86)\windows kits\10\include\10.0.19041.0\um\shobjidl_core.h(23789): error C2061: syntax error: identifier 'MSG'
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(46,5): error MSB3075: "C:\UE_4.24\Engine\Build\BatchFiles\Build.bat CityParkEditor Win64 DebugGame -Project="C:\CityPark\CityPark.uproject" -WaitMutex -FromMsBuild"

==================

mitchellspryn, 2021-03-04:
This sounds like your UE4 environment isn't set up properly. Can you build projects without AirSim? You can also try regenerating the solution by right-clicking on the .uproj file and clicking "Generate solution".

There also appear to be some extra files in your UrdfSim enlistment. For example, I see compilation failures in the following files in the first steps, but don't see them in the repo anywhere:
* C:/CityPark/Plugins/AirSim/Source/AirLib/include/unsupported/test/cxx11_tensor_reduction.cpp
* C:\CityPark\Plugins\AirSim\Source\AirLib\include\bench\tensors\tensor_benchmarks_sycl.cc

I cannot repro this on my end. I did the following:

* Download CityPark Environment
* Open the project in UE 4.24
* Add an empty C++ class to generate the solution
* Close the editor
* From a fresh clone of UrdfSim, build using 'build.cmd' from a developer command prompt. 
* Copy the output of `Unreal/plugins/AirSim` into `Plugins/Airsim` in the new project.
* Update the .uproj file to add the Airsim plugin
* Regenerate the solution. 
* Run solution in "Development." 
* Set default game mode to "AirSimGameMode"

Once I did that (and added in some of the lunabot assets), the game started as expected.

[image]

Since I've done the work to build this environment, I can release a binary drop in the releases page if it would help. It'll probably take some time (maybe days :( ) to fully build, as I don't have a very powerful computer.

guldamkwak3114, 2021-03-05:
Custom Environment Compile works for original Airsim. I followed exactly what you described. I have no idea yet why it produces errors. 

Could you upload the binary drop in the release so that I can test it in my machine? Thanks!

guldamkwak3114, 2021-03-05:
When I build URDFSim, it generates the following errors

"C:\UrdfSim\AirSim.sln" (default target) (1) ->
"C:\UrdfSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"C:\UrdfSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]
  c:\urdfsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cannot open include file: 'Eigen/Dense': No s
uch file or directory [C:\UrdfSim\AirLib\AirLib.vcxproj]

So I downloaded Eigen from 
https://eigen.tuxfamily.org/

and unzip, copy the all contents to C:\UrdfSim\AirLib\Include. Maybe it causes the error. It seems I need copy only the Eigen/Dense Folder to AirLib\Include Let me check.

guldamkwak3114, 2021-03-05:
I managed to compile CityParkEnvironment using URDFSim. It shows rendered car correctly, However, when I try to pawn my custom robot, which is rendered correctly in threebridged environment you realeased, it crashes and generated the following error 

[image]

LoginId:4731092a48a3dce4fc1e21870529d8b2
EpicAccountId:753e16548aa84831851e137911f7c640

Assertion failed: Pair != nullptr [File:C:\UE_4.24\Engine\Source\Runtime\Core\Public\Containers/Map.h] [Line: 566]

UE4Editor_Core
UE4Editor_Core
UE4Editor_AirSim!StaticMeshGenerator::CreateUnscaledMeshForLink() [c:\citypark\plugins\airsim\source\vehicles\urdfbot\meshgeneration\staticmeshgenerator.cpp:181]
UE4Editor_AirSim!AUrdfBotPawn::CreateLinkFromSpecification() [c:\citypark\plugins\airsim\source\vehicles\urdfbot\urdfbotpawn.cpp:367]
UE4Editor_AirSim!AUrdfBotPawn::ConstructFromFile() [c:\citypark\plugins\airsim\source\vehicles\urdfbot\urdfbotpawn.cpp:299]
UE4Editor_AirSim!AUrdfBotPawn::InitializeForBeginPlay() [c:\citypark\plugins\airsim\source\vehicles\urdfbot\urdfbotpawn.cpp:200]
UE4Editor_AirSim!ASimModeUrdfBot::setupVehiclesAndCamera() [c:\citypark\plugins\airsim\source\vehicles\urdfbot\simmodeurdfbot.cpp:179]
UE4Editor_AirSim!ASimModeUrdfBot::BeginPlay() [c:\citypark\plugins\airsim\source\vehicles\urdfbot\simmodeurdfbot.cpp:13]
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_AirSim!ASimHUD::createSimMode() [c:\citypark\plugins\airsim\source\simhud\simhud.cpp:311]
UE4Editor_AirSim!ASimHUD::BeginPlay() [c:\citypark\plugins\airsim\source\simhud\simhud.cpp:37]
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
kernel32
ntdll

Could you try have a look? Thanks!!

guldamkwak3114, 2021-03-05:
Lunabot Rendering also works! 

[image]

mitchellspryn, 2021-03-05:
Great. The binary drop finally finished uploading, but looks like you don't need it :).  https://github.com/mitchellspryn/UrdfSim/releases/tag/1.0.0. 

Now that you have the code running, you can place a breakpoint where it crashes. My guess is that something is misspelled in the URDF, or it's referencing an asset which isn't included in this environment. 

BTW, the lunabot looks a bit wonky (e.g. the bucket isn't straight). Do you have physics substepping enabled (top bar->edit->project settings->Engine->Physics->Substepping / Substepping Async boxes are checked)?

guldamkwak3114, 2021-03-05:
I conducted 
"top bar->edit->project settings->Engine->Physics->Substepping / Substepping Async boxes are checked)"

[image]

what do you mean by 
editor has focus ?

mitchellspryn, 2021-03-05:
There. That looks better. 

What I mean is that if you take focus away from the unreal editor (e.g. click on something else like a terminal), it will go into "energy saving" mode in which it uses less processing power. This can break the physics engine.

guldamkwak3114, 2021-03-05:
[image]
[image]

We managed to render our custom robot in out custom environment. Thanks!!

mitchellspryn, 2021-03-05:
Great! I'll close this issue and the related ones, then.

closed 2021-03-05 by mitchellspryn