mitchellspryn/UrdfSim #42: how  to  config  Eigen

kylo-ke, 2021-03-09:
hello，there have some error  when  i  excute Build.cmd 
fatal error C1083: Cannot open include file:
 'Eigen/Dense': No such file or directory [D:\A_UnrealEngine\UrdfSim-master\AirLib\AirLib.vcxproj]
.........

how to solve this problem?  thank you

InfraredLaser, 2021-03-16:
I second this issue. Unable to execute build.cmd due to this same error.
The error also states, "Cannot open include file: 'corecrt.h': No such file or directory"

Here is my current setup:
VS 2017 with UE 2.24.3

Is there a fix for this?
Thanks

mitchellspryn, 2021-03-17:
Interesting. I just tried a clean build, and it downloaded eigen properly. Perhaps try executing the download manually [as described in the steps here](https://github.com/mitchellspryn/UrdfSim/blob/master/build.cmd#L146). 

For the corecrt.h issue, sounds like you either have the wrong windows development kit installed or don't have it linked properly in your solution. See if you can build a generic UE4 c++ project without UrdfSim.

InfraredLaser, 2021-03-17:
Thanks for the response! 
I downloaded the latest eigen and placed it into the cloned UrdfSim repo folder and it seemed to solve the eigen issue when running build.cmd. 

Now for the "corect.h" error, I am using Windows 10 and under the build.cmd messages there is an output "Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363". What is the proper SDK to install so that this issue is removed?

Thanks

mitchellspryn, 2021-03-17:
```
See if you can build a generic UE4 c++ project without UrdfSim.
```

Did you do this? This sounds like a problem with your unreal install.

FWIW, the version I have installed has this manifest:

```
<FileList
  DisplayName = "Universal Windows"
  PlatformIdentity = "UAP, Version=10.0.17763.0"      
  TargetFramework = ".NETCore,version=v4.5.3;.NETFramework,version=v4.5.3"
  MinVSVersion = "14.0"
  MinOSVersion = "6.1"
  MaxOSVersionTested = "10.0"
  UnsupportedDowntarget = "Windows, version=8.1">

  <File Reference = "Windows">
    <ToolboxItems VSCategory = "Toolbox.Default"/>
  </File>
</FileList>

```

mitchellspryn, 2021-03-17:
I've gone ahead and updated the build script to use the new eigen location. Now it downloads properly.

closed 2021-03-17 by mitchellspryn

InfraredLaser, 2021-03-17:
Ok I found the solution to my issue with "corect.h".

I have not tried building a generic UE4 c++ project as I suspected that may have not been the issue with the build. Here is the solution that solves the "corect.h" issue:

Open Visual Studio Installer > Select "Modify" under visual studio professional 2017 > Individual Components > Compilers, build tools, and runtimes > Check "Windows Universal CRT SDK"

Run build.cmd and it should work.