mitchellspryn/UrdfSim #45: STL Format issues

henry-tra, 2021-03-30:
Hi, 
I used the SolidWorks to URDF Exporter in order to create the URDF with its STL files from some simple volume shapes (cylinder and box). I also created the XML file but after starting the Unreal Editor, I get the following error:

[image]

Please, can you help me?

quentingllmt, 2021-03-31:
Hi,
Maybe you'll find answers there :
https://github.com/mitchellspryn/UrdfSim/blob/master/docs/CustomMesh.md

henry-tra, 2021-04-01:
I have created a new STL file. The error is gone now but I can only see the coordiantes systems and now there is a further error:

[image]

Do I have to generate collision in order to see the shapes (with VHACD in Blender)?

closed 2021-04-01 by henry-tra

InfraredLaser, 2021-06-15:
Hi @henry-tra, 
I am currently having this same issue. What was your solution? I looked though the CustomMesh docs but I was still unable to fix the error.
Thanks

InfraredLaser, 2021-06-17:
> Hi @henry-tra,
> I am currently having this same issue. What was your solution? I looked though the CustomMesh docs but I was still unable to fix the error.
> Thanks

I was able to find a solution for this issue just in case anyone else comes into this issue.

1. I took a mesh that was designed in SolidWorks and had to save it as an STL.
2. I'm not sure if SolidWorks could save ASCII STLs, so I first saved each part as STLs. 
3. I imported the STL meshes into Blender to then export it as an ASCII STL.
3. Configuring the XML file by setting the paths to the location of these ASCII STLs then fixes this issue.