mitchellspryn/UrdfSim #49: RobotArm "base" mesh registers collisions even when Blacklisted.

InfraredLaser, 2021-05-20:
When running the example RobotArm python script, I noticed that the "base" mesh still registers collisions when the simulator is running. Under the .json file I noticed that this mesh should be blacklisted (code below), yet it sill registers collisions. The .json file that I ran is the default given in the example. 

Settings: Windows 10, UE 4.24.3, vs2017

```
"UrdfBot": {
          "VehicleType": "UrdfBot",
          "DebugSymbolScale": 5,
          "CollisionBlacklist": [
              {
                  "BotMesh": "base", 
                  "ExternalActorRegex": "Ground"
              }
          ],
```

mitchellspryn, 2021-05-20:
Which meshes is it registering collision with? Should they be matching the regex?

InfraredLaser, 2021-05-20:
Here is a screen shot of the detected collision.
[image]

I thought it was the base, but perhaps it could also be a different component?

mitchellspryn, 2021-05-20:
The collision is with mesh `BP_RobotArmStand_5`. That does not match the regex `Ground`. Updating the regex to match `BP_RobotArmStand_5` should stop the events from being registered.

InfraredLaser, 2021-05-20:
Thanks, now it works perfectly.

InfraredLaser, 2021-06-01:
When the "Player Start" object is rotated 180 degrees I noticed this issue persisting with the UrdfBot mesh. 

Here is an image of the issue:
[image]

It appears that the robot arm mesh does not like being rotated in order for the collisions to be blacklisted. Is there a way to add multiple objects to be blacklisted by regex?

mitchellspryn, 2021-06-02:
`CollisionBlacklist` is a list of objects, so you should be able to just add another one if you want to exclude more meshes from collision.

InfraredLaser, 2021-06-02:
Thanks for the reply @mitchellspryn, 

What would be the best way to do so? Here is a blacklist that I have so far:
```
"CollisionBlacklist": [
        {
          "BotMesh": "base",
          "ExternalActorRegex": "BP_RobotArmStand_5, UrdfBot"
        }
      ],
```
This does not seem to work when blocking the UrdfBot mesh.

mitchellspryn, 2021-06-02:
`BP_RobotArmStand_5, UrdfBot` is probably not the regex you want.

You probably want something like
```
{
  "BotMesh": "base", 
  "ExternalActorRegex": "BP_RobotArmStand_5"
},
{
  "BotMesh": "base",
  "ExternalActorRegex": "ObjID 0"
}
```

BTW, you should probably give that mesh you're colliding with a more descriptive name; it will make it easier to pick out.

InfraredLaser, 2021-06-03:
I tried the above blacklist settings and it seems that UrdfSim still registers the collision when the robot arm closes.
I also tried the below settings with the same conclusion. 
```
{
  "BotMesh": "base",
  "ExternalActorRegex": "UrdfBot"
}
```