mitchellspryn/UrdfSim #51: Unreal Process has crashed: Unhandled Exception: 0xe06d7363

InfraredLaser, 2021-06-07:
I'm getting this error when trying to boot up my UrdfSim environment from the Epic Games Launcher or through Visual Studio.
I started up the environment through the above processes and it worked before. All of the sudden the environment crashes when trying to load it.

Platform: Windows 10, vs2017, Unreal 2.24.3

**Below is the following error:**
LoginId:a434651d4fb6b6c0c52e0d9fed62d432
EpicAccountId:

Unhandled Exception: 0xe06d7363

UE4Editor_AirSim!nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer>::parser::expect() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\common_utils\json.hpp:11607]
UE4Editor_AirSim!nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer>::parser::parse_internal() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\common_utils\json.hpp:11461]
UE4Editor_AirSim!nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer>::parser::parse_internal() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\common_utils\json.hpp:11462]
UE4Editor_AirSim!nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer>::parser::parse_internal() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\common_utils\json.hpp:11462]
UE4Editor_AirSim!nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer>::parser::parse_internal() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\common_utils\json.hpp:11462]
UE4Editor_AirSim!nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer>::parser::parse() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\common_utils\json.hpp:11384]
UE4Editor_AirSim!msr::airlib::Settings::loadJSonString() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\airlib\include\common\settings.hpp:64]
UE4Editor_AirSim!ASimHUD::ASimHUD() [d:\unreal projects\factorysim_urdf\plugins\airsim\source\simhud\simhud.cpp:23]
UE4Editor_CoreUObject
UE4Editor_CoreUObject
UE4Editor_CoreUObject
UE4Editor_CoreUObject
UE4Editor_Core
UE4Editor_Core
UE4Editor_Projects
UE4Editor_Projects
UE4Editor_Projects
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
kernel32
ntdll

InfraredLaser, 2021-06-08:
Looking through the call stack the error appears in a file "json.hpp". There seems to be a parse error, but I am not quite sure what is causing this. I checked online forums for similar issues, some of the fixes result in removing meshes/placed objects in the map. However, before the error starts, the editor loads up to 71% and then crashes. 

@mitchellspryn any ideas on how I should go about fixing this issue?

Below is an image of the vs error.
[image]

mitchellspryn, 2021-06-08:
My guess is you have an invalid json. One potential step is to look at the json string being passed in `msr::airlib::Settings::loadJSonString`, and use a tool like [jsonlint](https://jsonlint.com/) to see if it's actually valid.

InfraredLaser, 2021-06-08:
Ah I found my error. Looking into my **_settings.json_** file I noticed that there was a syntax issue. Fixing the issue allowed the environment to load properly. Thanks for the help @mitchellspryn !

closed 2021-06-08 by InfraredLaser