mitchellspryn/UrdfSim #52: Materials for Custom STLs

InfraredLaser, 2021-07-08:
When trying to set the path for a custom material I noticed that it does not get applied to my custom STL mesh. When ends up showing is a default grey material. When looking through the issues I noticed that in #15 applying materials to custom STLs does not seem to work. Has this issue been fixed?

Here are the paths I used  to set up custom materials:
[image]

Here is a sample mesh that shows only the default grey material.
[image]

InfraredLaser, 2021-07-27:
@mitchellspryn 

I am trying a potential solution to this issue. Is it possible to add materials to custom ascii STLs if the STLs are first imported as static meshes in unreal engine, then linked using the xml file?

mitchellspryn, 2021-07-28:
Sorry, missed this the first time. 

If you can import it into unreal, that's definitely the way to go. In addition to being simpler at runtime, you'll get the benefits of pre-baked lighting for the mesh textures, so it will look better.

InfraredLaser, 2021-07-28:
Thanks @mitchellspryn ,

I tried to find the example to link the unreal meshes in the XML, but the only file that exists in the example folder is just the settings-mesh.json file. 
How can we link unreal meshes in the XML and what would the appropriate file path look like?

mitchellspryn, 2021-07-28:
[Try this](https://github.com/mitchellspryn/RoboMagellanOrchestrator/blob/master/RoboMagellanOrchestrator/rm_bot_client/RoboMagellanBot.xml#L38)

InfraredLaser, 2021-07-29:
The above method works. Importing the STL's to UE is the way to go. 
Additionally, using this method I am able to change the materials on my meshes though UE instead of the XML giving me greater flexibility over custom meshes.

closed 2021-07-29 by InfraredLaser