mitchellspryn/UrdfSim #54: simGetImages API does not accept image resolutions above 256x144

InfraredLaser, 2021-09-01:
When playing around with the image API I noticed that resolutions above 256x144 would not be saved or its image data not collected. When debugging it seems that simGetImages is only able to grab images at the default 256x144 resolution. Newer versions of AirSim resolved this issue, I'm hoping there could be a quick fix for this version of UrdfSim.

InfraredLaser, 2021-09-09:
@mitchellspryn can the imaging API's be updated to resolve this issue?

mitchellspryn, 2021-09-09:
I'm sure they can be :). I am currently busy with a bunch of other projects, and won't be able to take a look at this issue for some time.

InfraredLaser, 2021-09-09:
The newer version of AirSim is capable of taking higher resolution images. I tried replacing the newer AirSim API's with UrdfSim's version, but I could not get the API's to work.

mitchellspryn, 2021-09-09:
Yes, the two code bases have diverged quite a bit. So I would expect that there would need to be some engineering effort involved to get it right.

InfraredLaser, 2021-11-12:
@mitchellspryn 

So I had a further looked into this issue. It turns out that no matter how the camera defaults in _settings.json_ are changed, UrdfSim will only take their default values. I'm thinking that if we find the location in the UrdfSim plugin which transfers the settings in _settings.json_ to the PythonClient we can potentially resolve this issue.