mitchellspryn/UrdfSim #58: Unable to open or build default Blocks.uproject from Airsim.

yashmewada9618, 2023-01-31:
I am a bit new to Airsim and URDFsim. My goal is to import a robot urdf file in Airsim/URDFsim and simulate tasks like autonomous navigation using image segmentation. For testing purposes, I built Airsim on my machine, and after million tries and debugging I was able to open up the default project. Now to move ahead in this project I want to import the URDF file of the robot, so I did was imported the Blocks. project file from Airsim and ran the below command to build it, (*Note - I was not able to build this from UEeditor itself, hence manual building.)
```console
UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/bin/mono UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Linux Development -Project=/UrdfSim/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress
```
After solving a few of the case-sensitive errors, I am now stuck at the below error.

```console
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:86:63: error: converting the enum constant to a boolean [-Werror,-Wint-in-bool-context]
    MayLinearVectorize = bool(MightVectorize) && MayLinearize && DstHasDirectAccess
                                                              ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:607:20: note: in instantiation of template class 'Eigen::internal::copy_using_evaluator_traits<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::assign_op<double, double>>' requested here
  typedef typename AssignmentTraits::PacketType PacketType;
                   ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:722:10: note: in instantiation of template class 'Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::assign_op<double, double>, 0>' requested here
  Kernel kernel(dstEvaluator, srcEvaluator, func, dst.const_cast_derived());
         ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:862:5: note: in instantiation of function template specialization 'Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double>>' requested here
    call_dense_assignment_loop(dst, src, func);
    ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:819:46: note: in instantiation of member function 'Eigen::internal::Assignment<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double>, Eigen::internal::Dense2Dense, void>::run' requested here
  Assignment<ActualDstTypeCleaned,Src,Func>::run(actualDst, src, func);
                                             ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/PlainObjectBase.h:728:17: note: in instantiation of function template specialization 'Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double>>' requested here
      internal::call_assignment_no_alias(this->derived(), other.derived(), internal::assign_op<Scalar,typename OtherDerived::Scalar>());
                ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/Matrix.h:278:15: note: in instantiation of function template specialization 'Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3>>::_set_noalias<Eigen::Matrix<double, 3, 3, 0, 3, 3>>' requested here
        Base::_set_noalias(other);
              ^
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/GeodeticConverter.hpp:199:12: note: in instantiation of member function 'Eigen::Matrix<double, 3, 3, 0, 3, 3>::Matrix' requested here
    return ret;
           ^
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:2:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirBlueprintLib.gen.cpp:8:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:22:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:12:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/CommonStructs.hpp:7:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:12:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/Dense:1:
In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/Core:415:
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:86:63: error: converting the enum constant to a boolean [-Werror,-Wint-in-bool-context]
    MayLinearVectorize = bool(MightVectorize) && MayLinearize && DstHasDirectAccess

```
More precisely the error is ...
```console
/home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:86:63: error: converting the enum constant to a boolean [-Werror,-Wint-in-bool-context]
```

I am unable to trace back the error. I might be missing something or doing something wrong here.

### System setup
****
> 1. OS installed on the PC?
>    
>    * [x]  Ubuntu 20.04 LTS (Focal Fossa)
>2. UE4 version?
>
>     * [x]  UE4.27

*Also I am running this on a separate HDD and not my laptop due to space constraints.
Some other resources or help regarding importing a URDF file in Airsim/URDFsim project will be deeply appreciated.

Vaibhav2181, 2023-06-21:
Hello, 

> I am a bit new to Airsim and URDFsim. My goal is to import a robot urdf file in Airsim/URDFsim and simulate tasks like autonomous navigation using image segmentation. For testing purposes, I built Airsim on my machine, and after million tries and debugging I was able to open up the default project. Now to move ahead in this project I want to import the URDF file of the robot, so I did was imported the Blocks. project file from Airsim and ran the below command to build it, (*Note - I was not able to build this from UEeditor itself, hence manual building.)
> 
> ```
> UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/bin/mono UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Linux Development -Project=/UrdfSim/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress
> ```
> 
> After solving a few of the case-sensitive errors, I am now stuck at the below error.
> 
> ```
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:86:63: error: converting the enum constant to a boolean [-Werror,-Wint-in-bool-context]
>     MayLinearVectorize = bool(MightVectorize) && MayLinearize && DstHasDirectAccess
>                                                               ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:607:20: note: in instantiation of template class 'Eigen::internal::copy_using_evaluator_traits<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::assign_op<double, double>>' requested here
>   typedef typename AssignmentTraits::PacketType PacketType;
>                    ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:722:10: note: in instantiation of template class 'Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3>>, Eigen::internal::assign_op<double, double>, 0>' requested here
>   Kernel kernel(dstEvaluator, srcEvaluator, func, dst.const_cast_derived());
>          ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:862:5: note: in instantiation of function template specialization 'Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double>>' requested here
>     call_dense_assignment_loop(dst, src, func);
>     ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/AssignEvaluator.h:819:46: note: in instantiation of member function 'Eigen::internal::Assignment<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double>, Eigen::internal::Dense2Dense, void>::run' requested here
>   Assignment<ActualDstTypeCleaned,Src,Func>::run(actualDst, src, func);
>                                              ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/PlainObjectBase.h:728:17: note: in instantiation of function template specialization 'Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::internal::assign_op<double, double>>' requested here
>       internal::call_assignment_no_alias(this->derived(), other.derived(), internal::assign_op<Scalar,typename OtherDerived::Scalar>());
>                 ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3/Eigen/src/Core/Matrix.h:278:15: note: in instantiation of function template specialization 'Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3>>::_set_noalias<Eigen::Matrix<double, 3, 3, 0, 3, 3>>' requested here
>         Base::_set_noalias(other);
>               ^
> /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/GeodeticConverter.hpp:199:12: note: in instantiation of member function 'Eigen::Matrix<double, 3, 3, 0, 3, 3>::Matrix' requested here
>     return ret;
>            ^
> In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:2:
> In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirBlueprintLib.gen.cpp:8:
> In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:22:
> In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:12:
> In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/CommonStructs.hpp:7:
> In file included from /home/yash/mountpoint/ALLAN/UrdfSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
[... cut ...]
> *Also I am running this on a separate HDD and not my laptop due to space constraints. Some other resources or help regarding importing a URDF file in Airsim/URDFsim project will be deeply appreciated.

Hello, 
 If I have tried this solution and it worked for me.
Changes ->
PATH_TO_AIRSIM/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/CommonStructs.hpp:272:31:
 error: converting the result of '<<' to a boolean; did you mean '(1 << index) != 0'? [-Werror,-Wint-in-bool-context]
        return switches && (1 << index) ? 1 : 0;

to

      return switches && ((1 << index) != 0) ? 1 : 0;

ssaigarimella, 2024-11-14:
@yashmewada9618 @Vaibhav2181 how did you get a blocks environment from UrdfSim? as in the Blocks.uproject itself, it is missing in the repo